bulletphysics/bullet3 #1322: Could NOT find PythonLibs (missing:  PYTHON_LIBRARIES) (found suitable exact version "2.7.12")

wangROS, 2017-09-18:
Could NOT find PythonLibs (missing:  PYTHON_LIBRARIES) (found suitable exact version "2.7.12")

erwincoumans, 2017-09-18:
cmake's find mechanism 'find' doesn't work well.  See also pull request 1131 (https://github.com/bulletphysics/bullet3/pull/1131)
You can use "pip install -e ." in the root of Bullet, or simply use pip install pybullet for now.

ahundt, 2017-09-19:
The correct command line parameters need to be specified. If, for example, a path to the python interpreter and the python library are found and the versions do not match exactly the cmake script will correctly report an error. Here is the most recent command I built with via CMake, but it was some time ago so I can't say it is perfect and working on master, but it might help figure out what to do.

```
cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/bullet/2.86.1_1 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_VERBOSE_MAKEFILE=ON -Wno-dev -DHAVE_CLOCK_GETTIME:INTERNAL=0 -DINSTALL_EXTRA_LIBS=ON -DBUILD_UNIT_TESTS=OFF -DUSE_DOUBLE_PRECISION=ON -DBUILD_BULLET2_DEMOS=ON -DBUILD_PYBULLET_MAC_USE_PYTHON_FRAMEWORK=ON -DBUILD_PYBULLET=ON -DBUILD_PYBULLET_NUMPY=ON -DBUILD_PYBULLET_CLSOCKET=ON -DBUILD_PYBULLET_ENET=ON -DPYTHON_EXECUTABLE=/usr/local/opt/python/bin/python -DPYTHON_LIBRARIES=/usr/local/Cellar/python/2.7.13/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib -DPYTHON_INCLUDE_DIRS=/usr/local/Cellar/python/2.7.13/Frameworks/Python.framework/Versions/2.7/include/python2.7 -DPYTHON_NUMPY_INCLUDE_DIR=/usr/local/lib/python2.7/site-packages/numpy/core/include -DBUILD_ENET=ON -DBUILD_CLSOCKET=ON -DBUILD_SHARED_LIBS=ON -DFRAMEWORK=OFF -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/bullet/2.86.1_1/Frameworks -DCMAKE_INSTALL_NAME_DIR=/usr/local/Cellar/bullet/2.86.1_1/Frameworks
```

erwincoumans, 2017-09-19:
There must be some sane find python cmake script that automates all this, without having to specify it, right? Having to tell cmake exactly all those details defeats the purpose of a find Python script.

ahundt, 2017-09-19:
> There must be some sane find python cmake script out there that automates all this, without having to specify it, right? Having to tell cmake exactly all those details defeats the purpose of a find Python script.

I sure hope so... I incorporated the best one I could find with a compatible license.

erwincoumans, 2017-09-19:
Also, the message is really odd:
>> found suitable exact version "2.7.12"

So why doesn't it find this suitable exact version, if it found it?

erwincoumans, 2017-09-19:
I'll try to re-enable building of python in travis here, let's see what the error is:
https://github.com/bulletphysics/bullet3/pull/1323

erwincoumans, 2017-09-19:
>>I incorporated the best one I could find with a compatible license.

yes, I didn't mean to blame you. It is frustrating that finding Python is such a pain with cmake. That's why I moved to using pip/setuptools, at least that seems to work smooth.

erwincoumans, 2017-09-19:
If we cannot make the travis work, I will revert to the old way of finding python.

ahundt, 2017-09-21:
Here is a second example in case that helps, this is for mac so obviously the paths would need to be updated for linux and the mac only flags would need to be removed:

```
± cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/bullet/HEAD-47eae8d_1 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_VERBOSE_MAKEFILE=ON -Wno-dev -DHAVE_CLOCK_GETTIME:INTERNAL=0 -DINSTALL_EXTRA_LIBS=ON -DBUILD_UNIT_TESTS=OFF -DUSE_DOUBLE_PRECISION=ON -DBUILD_PYBULLET_MAC_USE_PYTHON_FRAMEWORK=ON -DBUILD_PYBULLET=ON -DBUILD_PYBULLET_MAC_USE_PYTHON_FRAMEWORK=ON -DBUILD_PYBULLET_NUMPY=ON -DBUILD_PYBULLET_CLSOCKET=ON -DBUILD_PYBULLET_ENET=ON -DBUILD_ENET=ON -DBUILD_CLSOCKET=ON -DBUILD_SHARED_LIBS=OFF -DBUILD_BULLET2_DEMOS=OFF -DPYTHON_NUMPY_INCLUDE_DIR=/usr/local/Cellar/python/2.7.13/Frameworks/Python.framework/Versions/2.7/include/python2.7
```

erwincoumans, 2017-09-21:
Having to specify where python is defeats the purpose of a 'find' script, unless you have multiple versions of Python. Travis machines only have 1 Python version by default. Luckily the 'find_package' finds Python libs and include paths just fine on Travis and that's important to make sure things are build properly. 

Wouldn't the current 'find' python script work fine if you provide all variables explicitly?

ahundt, 2017-09-22:
Oh wait, all my configuration code that found specific versions and set things up consistently was removed in https://github.com/bulletphysics/bullet3/commit/03cde2ee9bed76c186a1507be6319328c3bf728c, in that case it makes perfect sense that things don't work!

What happened to the changes in https://github.com/bulletphysics/bullet3/pull/1131?

ahundt, 2017-09-22:
fixed in #1331

PR #1331 (2017-09-22): Cmake pybullet fix
fix #1322, the problem is that python itself declares its own lib file to be in the wrong location in certain linux versions. Here I primarily add the extra missing path component to the set of search paths. I also restore changes from previous fix attempts that were incorrect.

https://github.com/bulletphysics/bullet3/pull/1131 is a past change of interest to this one
closed 2017-09-22 by erwincoumans