bulletphysics/bullet3 #1324: stl rendered very different in main view vs getCameraImage

jtoy, 2017-09-19:
Not sure if this is considered expected behaviour or is this a bug? I load a key object as a URDF wrapped around an  stl file and it renders as a large block.  When I use getCameraImage to take a picture it renders all the details though and not a large block: https://cl.ly/3b2E1e0I3D2M
Does the different  rendering mess with the collisions at all? What is going on here?

And here is what a generic stl renderer renders it as: https://cl.ly/3o0B1Y2F3a0z

jtoy, 2017-09-24:
I tested this with a bunch of stl files. is there a way to tell the OpenGL renderer to render a full fidelity shape instead of a lower resolution version? I'm assuming the renderer doesnt show the full fidelity for computational cost reasons, is that correct?

erwincoumans, 2017-09-24:
The OpenGL renderer shows the collision shapes, which is the convex hull of the STL file. The TinyRenderer takes the original triangle mesh. This is not due to rendering performance reasons. If you want to use a visual shape, it is best to use OBJ files. It is a bit complicated: by default, Bullet will take the convex hull of shapes. If you want to have concave shapes, you need to apply convex decomposition (breaking the concave shape into a collection of convex shape). You can use VHACD for this, in the Bullet/Extras folder.

closed 2019-05-30 by erwincoumans