bulletphysics/bullet3 #1326: is OPENGL renderer taking into account lightDirection, lightColor,...?

FernandoCamaro, 2017-09-20:
I am not sure if this is an expected behavior but the rendered images with OPENGL don't seem to
be afected by lightColor, lightDirection,...
Here are the two images generated from the following script. Both images have the same
configuration, the only difference is the change in the renderer.
One thing that is very clear is the difference in the shadow of the big red cube. In one image it appears in the right side while in the other appears in the left side.

 
`import matplotlib.pyplot as plt`
`import pybullet as p`
`import pybullet_data`
`physicsClient=p.connect(p.GUI)`
`p.setAdditionalSearchPath(pybullet_data.getDataPath())`
`planeId=p.loadURDF("plane.urdf")`
`cubeStartOrientation = p.getQuaternionFromEuler([0,0,0])`
`boxId=p.loadURDF("cube_small.urdf",[0,0,1],cubeStartOrientation,globalScaling=10)`
`boxId2=p.loadURDF("cube_small.urdf",[1,1,0],cubeStartOrientation,globalScaling=5)`
`viewMatrix=p.computeViewMatrix([2,2,1],[0,0,0],[0,0,1])`
`projMatrix=p.computeProjectionMatrixFOV(fov=90,aspect=4./3.,nearVal=0.01,farVal=100)`
`p.changeVisualShape(boxId,-1,rgbaColor=[1,0,0,1])`
`width, height, rgb1, depth,seg = p.getCameraImage(1024,768,viewMatrix,projMatrix, shadow=1, lightDirection=[-1,-20,10],lightAmbientCoeff=0.5,lightDistance=100,lightColor=[1,1,1],lightDiffuseCoeff=0.4,lightSpecularCoeff=0.1,renderer=p.ER_TINY_RENDERER)`
`width, height, rgb2, depth,seg = p.getCameraImage(1024,768,viewMatrix,projMatrix, shadow=1, lightDirection=[-1,-20,10],lightAmbientCoeff=0.5,lightDistance=100,lightColor=[1,1,1],lightDiffuseCoeff=0.4,lightSpecularCoeff=0.1,renderer=p.ER_BULLET_HARDWARE_OPENGL)`
`plt.imshow(rgb1[:,:,0:3])`
`plt.show()`
`plt.imshow(rgb2[:,:,0:3])`
`plt.show()`

Image obtained with tiny renderer
[image]

image obtained with opengl renderer
[image]

erwincoumans, 2017-10-11:
It could be that not all parameters are passed to the OpenGL renderer, but only to TinyRenderer. I don't have time right now to focus on this.

luator, 2021-02-26:
Are there any news to this?

closed 2021-04-26 by erwincoumans