bulletphysics/bullet3 #1329: Please rename identifer _

nicolaichuk, 2017-09-21:
please rename identifer '_' in this type 

- [btSoftBody::eAeroModel::_](https://github.com/bulletphysics/bullet3/blob/master/src/BulletSoftBody/btSoftBody.h#L84)
- [btSoftBody::eVSolver::_](https://github.com/bulletphysics/bullet3/blob/master/src/BulletSoftBody/btSoftBody.h#L96)
- [btSoftBody::ePSolver::_](https://github.com/bulletphysics/bullet3/blob/master/src/BulletSoftBody/btSoftBody.h#L102)
- [btSoftBody::eSolverPresets::_](https://github.com/bulletphysics/bullet3/blob/master/src/BulletSoftBody/btSoftBody.h#L111)
- [btSoftBody::eFeature::_](https://github.com/bulletphysics/bullet3/blob/master/src/BulletSoftBody/btSoftBody.h#L119)
- [btSoftBody::fCollision::_](https://github.com/bulletphysics/bullet3/blob/master/src/BulletSoftBody/btSoftBody.h#L136)
- [btSoftBody::fMaterial::_](https://github.com/bulletphysics/bullet3/blob/master/src/BulletSoftBody/btSoftBody.h#L151)
- [btSoftBody::Joint::eType::_](https://github.com/bulletphysics/bullet3/blob/master/src/BulletSoftBody/btSoftBody.h#L489)

only-a-ptr, 2017-09-27:
Hello, random visitor here.

@nicolaichuk I'm wondering exactly would you like those renamed (besides being ugly) and what would be an acceptable name (given the `struct eMyEnum { enum Whatever {}; };` style is preserved)?

nicolaichuk, 2017-09-28:
@only-a-ptr

> I'm wondering exactly would you like those renamed

use c++11 enum classes  )