bulletphysics/bullet3 #1330: Heap Corruption on World Instantiation

thomasburbeck, 2017-09-21:
I've downloaded the latest Bullet source and I'm getting heap corruption.

This occurs only after instansiating a "world" object. I'm specifically trying to use "btDiscreteDynamicsWorld" as per the wiki tutorials but this also happens when using other "World" type objects I've found in other tutorials too. 

Finding it hard to track down any more useful info that that. My solution is currently stopped on a writeAccessViolation on line 94 of "btAlignedObjectArray"; but this could be a red herring as I get errors all over the place. Also errors have included "Access violation reading location 0x0000000000000000." and "Critical error detected c0000374".

thomasburbeck, 2017-09-21:
Attempting now to downgrade to a "Release" version as I'm sure this is probably an issue that is only occuring in the latest code.

thomasburbeck, 2017-09-21:
I am getting the same issue on 2.86.1. But I have just noticed the issue is intermittent; I ran it 5 times and got 4 different errors; but the 5th time my application ran. Something multithreaded?

iSLC, 2017-09-21:
Are you by any chance compiling with SIMD enabled and on a GCC/MinGW compiler?

erwincoumans, 2017-09-21:
Can you reproduce it with the default unmodified ExampleBrowser and the default unmodified premake and/or cmake build system that ships with Bullet? If so, please provide the steps.

thomasburbeck, 2017-09-21:
Compiling on VS2017 with /Od so I don't think SIMD should be used (according to my understanding). 

Currently unable to reproduce using the ExampleBrowser. Which obviously probably indicates something my end. I've only followed the wiki tutorial though, is it up to date? All I've done is the following:

1) Run build_visual_studio_vr_pybullet_double.bat
2) Open the VS2010 solution and auto-upgrade to VS2017
3) Open my Solution & Add the Existing Projects for: BulletCollision, BulletDynamics and Linear Math
4) Added the source code from the Hello World tutorial found here: http://bulletphysics.org/mediawiki-1.5.8/index.php/Hello_World
5) Changed Compilation of the three Bullet Libraries listed above to be /MDd rather than /MTd. 
6) Compile, run, heap corruption.

And I can reproduce that following the steps above. 

P.S. On a side note, doesn't affect me but, I did just get a "Vector Subscript out of range." assertion error in the pendulum example by changing the sliders value's; can't reproduce a second time.

thomasburbeck, 2017-09-21:
Oh I think I've found the problem, I think I was missing the following preprocessor defs:

BT_USE_DOUBLE_PRECISION
_DEBUG=1

Obviously my fault as I should have looked in more detail and found this before now. But, perhaps worth adding these to the doc's? Just glancing at the tutorials though I don't see this in them, I've only figured this out from looking over the project properties. 

Thanks for taking the time to read this guys!

closed 2017-09-21 by thomasburbeck

erwincoumans, 2017-09-21:
Good you sorted it. It would probably be a good thing to make the single precision build and double precision build incompatible, to force a linker error, instead of some cryptic run-time error. Not sure yet how to do this.

pisuto, 2018-09-13:
sorry, i meet the same problem on ubuntu14.04
can you explain the solution in detail?
thank you.

jiangcaiyang, 2020-08-17:
I meet the problem too. My approach is to wrap default classes with 16bit aligned. This is nothing to do with BT_USE_DOUBLE_PRECISION.

Test code:
```
#include <iostream>

#include "BulletCollision/CollisionDispatch/btDefaultCollisionConfiguration.h"
#include "BulletCollision/CollisionDispatch/btCollisionDispatcher.h"

ATTRIBUTE_ALIGNED16(class) CollisionDispatcher: public btCollisionDispatcher
{
public:
    CollisionDispatcher(btCollisionConfiguration* collisionConfiguration):
        btCollisionDispatcher(collisionConfiguration)
    {

    }
};

ATTRIBUTE_ALIGNED16(class) DefaultCollisionConfiguration: public btDefaultCollisionConfiguration
{

};

int main()
{
    std::cout << "Hello World!\n";

    DefaultCollisionConfiguration* conf = new DefaultCollisionConfiguration;
    CollisionDispatcher* dispatcher = new CollisionDispatcher(conf);
    delete dispatcher;
    delete conf;
}
```

erwincoumans, 2020-08-17:
As with previous person: Can you reproduce it with the default unmodified ExampleBrowser and the default unmodified premake and/or cmake build system that ships with Bullet? If so, please provide the steps. Bullet doesn't use multithreading using the default settings.

jiangcaiyang, 2020-08-29:
I use cmake to build bullet. I don't do anything special with bullet.
Here is my cmake command line:

cmake -S %bulletSourceDir% -B %buildFolder% -G "Visual Studio 16" ^
-DCMAKE_BUILD_TYPE=Release ^
-DBUILD_SHARED_LIBS=ON ^
-DBUILD_BULLET3=OFF ^
-DBUILD_CPU_DEMOS=OFF ^
-DBUILD_OPENGL3_DEMOS=OFF ^
-DBUILD_BULLET2_DEMOS=OFF ^
-DBUILD_UNIT_TESTS=OFF ^
-DCMAKE_EXE_LINKER_FLAGS_RELEASE="/DEBUG /INCREMENTAL:NO" ^
-DCMAKE_SHARED_LINKER_FLAGS_RELEASE="/DEBUG /INCREMENTAL:NO" ^
-DCMAKE_MODULE_LINKER_FLAGS_RELEASE="/DEBUG /INCREMENTAL:NO" ^
-DUSE_DOUBLE_PRECISION=OFF ^
-DCMAKE_INSTALL_PREFIX=%installDir%

I found 
```
CollisionDispatcher* dispatcher = new CollisionDispatcher(conf);
delete dispatcher;
```
OK, whereas
```
btCollisionDispatcher* dispatcher = new btCollisionDispatcher(conf);
delete dispatcher;
```
crashed.

I believe "ExampleBrowser" is somehow okay, unless there is some special flag set, the above code will crash.