bulletphysics/bullet3 #1347: Compilation problem on macOS High Sierra with -march=core2

scpeters, 2017-09-28:
They haven't been able to build a bottle for bullet yet on macOS 10.13 High Sierra due to a [compilation problem with b3GjkPairDetector.cpp when compiling for a core2 cpu architecture](https://github.com/Homebrew/homebrew-core/issues/18493#issuecomment-332764454). The compilation problem seems to trigger with `-O2` and higher, but is ok with `-O1` and `-O0`. On my machine, it works fine with `-march=native`, but homebrew bottles are built to be compatible with older architectures.

This might be a clang bug in XCode 9, but I figured I'd report it here first. Here's a few compiler commands for reproducing the problem, assuming you start from the root of the bullet source tree:

~~~
$ c++ --version
Apple LLVM version 9.0.0 (clang-900.0.37)
Target: x86_64-apple-darwin17.0.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
~~~

## Compiler flags that work

### -O3 -DNDEBUG -march=native

~~~
/Library/Developer/CommandLineTools/usr/bin/c++ \
  -I./src \
  -o b3GjkPairDetector.o \
  -c ./src/Bullet3OpenCL/NarrowphaseCollision/b3GjkPairDetector.cpp \
  -O3 -DNDEBUG -march=native
~~~

### -O1 -DNDEBUG -march=core2

~~~
/Library/Developer/CommandLineTools/usr/bin/c++ \
  -I./src \
  -o b3GjkPairDetector.o \
  -c ./src/Bullet3OpenCL/NarrowphaseCollision/b3GjkPairDetector.cpp \
  -O1 -DNDEBUG -march=core2
~~~

### -O0 -DNDEBUG -march=core2

~~~
/Library/Developer/CommandLineTools/usr/bin/c++ \
  -I./src \
  -o b3GjkPairDetector.o \
  -c ./src/Bullet3OpenCL/NarrowphaseCollision/b3GjkPairDetector.cpp \
  -O0 -DNDEBUG -march=core2
~~~

## Compiler flags that don't work

### -O3 -DNDEBUG -march=core2

~~~
/Library/Developer/CommandLineTools/usr/bin/c++ \
  -I./src \
  -o b3GjkPairDetector.o \
  -c ./src/Bullet3OpenCL/NarrowphaseCollision/b3GjkPairDetector.cpp \
  -O3 -DNDEBUG -march=core2
~~~

### -O2 -DNDEBUG -march=core2

~~~
/Library/Developer/CommandLineTools/usr/bin/c++ \
  -I./src \
  -o b3GjkPairDetector.o \
  -c ./src/Bullet3OpenCL/NarrowphaseCollision/b3GjkPairDetector.cpp \
  -O2 -DNDEBUG -march=core2
~~~

erwincoumans, 2017-09-28:
Thanks for reporting Stephen!
I'll look into it. Also, I will tag a new Bullet version soon, so please report any issues with current github master version.

erwincoumans, 2017-09-28:
What error do you get? It just hangs compiling? if so, it seems a bug in the clang/gcc compiler, right?

scpeters, 2017-09-28:
Thanks Erwin. We have a mac mini here with high sierra so I can assist with testing.

scpeters, 2017-09-28:
Yeah, the compiler just hangs. It does seem like a bug in clang to me

scpeters, 2017-09-28:
I can figure out how to report it to clang if you want, but I thought I'd show it to you first to see if you had any ideas.

erwincoumans, 2017-09-28:
That file (./src/Bullet3OpenCL/NarrowphaseCollision/b3GjkPairDetector.cpp)
is not really used, so we could just add #ifdef it out entirely, or delete
the file (it was from the OpenCL3 branch, and the file isn't use on
GPU/OpenCL and nehter on CPU (where we are still stuck with BulletCollision
(2.x).

Still, it would be good to report it to clang indeed.

scpeters, 2017-09-28:
I looked into it a tiny bit more, and one of the differences between `-O1` and `-O2` in clang is `-vectorize-loops`. I figured out how to remove that option from the clang invocation and it seems to work. I'll keep you updated as I work on reporting this to clang.

erwincoumans, 2017-09-28:
This old cruft code (b3GjkPairDetector) needs to be removed anyway, but it could be a good reproduction case for clang compiler issue indeed. For now, I will delete the file from the repo and let you know.

scpeters, 2017-09-28:
I was mistaken that `-vectorize-loops` is the source of the problem.

erwincoumans, 2017-09-29:
I removed the file, and will tag a new release very soon. It will become just a clang issue then.
https://github.com/bulletphysics/bullet3/pull/1348
Thanks for the report!

erwincoumans, 2017-09-29:
ok, I also tagged the 2.87 release: https://github.com/bulletphysics/bullet3/releases/tag/2.87

erwincoumans, 2017-09-29:
Hopefully the pybullet Python bindings will become more widely used too: it is as simple as "sudo pip install pybullet". Also, we could start considering using the Bullet C-API in Gazebo, for better multibody simulation support.

closed 2017-09-29 by erwincoumans

scpeters, 2017-10-03:
I started stripping out code to see how little was required to reproduce the clang error. I removed several hundred lines of code in https://github.com/scpeters/bullet3/commit/d6141a0f10034127c900fe28d5fa5ef3e2575f84

It may be possible to do more, but this seems to be the crux of code that illustrates the problem

* https://github.com/scpeters/bullet3/blob/d6141a0f10034127c900fe28d5fa5ef3e2575f84/src/Bullet3OpenCL/NarrowphaseCollision/b3GjkPairDetector.cpp

scpeters, 2017-10-04:
It looks like this bug has been fixed upstream:

* http://lists.llvm.org/pipermail/cfe-dev/2017-October/055664.html

erwincoumans, 2017-10-04:
That great. I'm not going to recover the file, good riddance.