bulletphysics/bullet3 #1351: How to create multiple revolute joints for a single body (properly)

cprogrammer1994, 2017-10-02:
Hello,

The [humanoid_symmetric_no_ground.xml](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_data/mjcf/humanoid_symmetric_no_ground.xml) file has some similar lines to the following:

```
<joint armature="0.01" axis="-1 0 0" name="left_hip_x" pos="0 0 0" ... type="hinge"/>
<joint armature="0.01" axis="0 0 -1" name="left_hip_z" pos="0 0 0" ... type="hinge"/>
<joint armature="0.01" axis="0 1 0" name="left_hip_y" pos="0 0 0" ... type="hinge"/>
<geom fromto="0 0 0 0 -0.01 -.34" name="left_thigh1" size="0.06" type="capsule"/>
<body name="left_shin" pos="0 -0.01 -0.403">
```

Please help me understand how could I create something similar using a `btMultiBody`.

I have tried to create "imaginary" links, but I am not sure how to set the mass and the inertia properly.

I did the following:

- create a btMultiBody with 3 links (fixedBase=true)
- create a btBoxShape.
- call setupRevolute 3 times.
- set the m_collider of the last link.

What should be the mass and the inertia of the intermediate links?

Thank you!

erwincoumans, 2017-10-02:
The MJCF importer code creates a btMultiBody with the intermediate links, you could try to re-use it?

The code is in examples\Importers\ImportMJCFDemo and ImportURDFDemo.

cprogrammer1994, 2017-10-02:
I did not know that mass = 0 and local inertia = (0,0,0) are valid values. I will change my code to use these values.
Thank you for the response!
Great project by the way. I really enjoy working with it.

closed 2017-10-02 by cprogrammer1994