bulletphysics/bullet3 #1352: doubt regarding btCompund shape

srajangarg, 2017-10-02:
I modified the `SimpleBox.cpp` example to test out compound shapes.

```
{
	btCompoundShape* compoundShape = new btCompoundShape();
	btBoxShape* cube  = new btBoxShape(btVector3(0.5, 0.5, 0.5));
	m_collisionShapes.push_back(cube);
	btTransform transform;
	
	// for (int i = 0; i < CUBES; i++)
	// {
	transform.setIdentity();
	transform.setOrigin(btVector3(0, 0, 0));
	compoundShape->addChildShape(transform, cube);

	transform.setIdentity();
	transform.setOrigin(btVector3(0, 1, 0));
	compoundShape->addChildShape(transform, cube);

	transform.setIdentity();
	transform.setOrigin(btVector3(0, 2, 0));
	compoundShape->addChildShape(transform, cube);

	transform.setIdentity();
	transform.setOrigin(btVector3(0, 1, 1));
	compoundShape->addChildShape(transform, cube);

	m_collisionShapes.push_back(compoundShape);
	transform.setIdentity();
	transform.setOrigin(btVector3(0, 10, 0));
	createRigidBody(1.0 , transform, compoundShape);
}
```
^ is what i added, apart from the existing ground shape.

It render properly, but the physics are off 
[image]

I cannot have it rest on diagonally on the ground. I push it (using my mouse), it behaves as if it is hinged and rises pack to this upright position. Can someone point out what I am doing wrong?

erwincoumans, 2017-10-02:
The btRigidBody is aligned with the center of mass. You need to recompute the shift to make sure the compound shape is re-aligned. Here is a snippet, we should have this as a basic example indeed.

`
	{
		btCompoundShape* compoundShape = new btCompoundShape();
		btBoxShape* cube  = new btBoxShape(btVector3(0.5, 0.5, 0.5));
		m_collisionShapes.push_back(cube);
		btTransform transform;
	
		transform.setIdentity();
		transform.setOrigin(btVector3(0, 0, 0));
		compoundShape->addChildShape(transform, cube);

		transform.setIdentity();
		transform.setOrigin(btVector3(0, 1, 0));
		compoundShape->addChildShape(transform, cube);

		transform.setIdentity();
		transform.setOrigin(btVector3(0, 2, 0));
		compoundShape->addChildShape(transform, cube);

		transform.setIdentity();
		transform.setOrigin(btVector3(0, 1, 1));
		compoundShape->addChildShape(transform, cube);
		
		transform.setIdentity();
		transform.setOrigin(btVector3(0, 10, 0));
		btScalar masses[4]={1,1,1,1};
		btTransform principal;
		btVector3 inertia;
		compoundShape->calculatePrincipalAxisTransform(masses,principal,inertia);
		btCompoundShape* compound2 = new btCompoundShape();
		m_collisionShapes.push_back(compound2);
#if 0
		//compound2->addChildShape(principal.inverse(), compoundShape);
#else	
		for (int i=0;i<compoundShape->getNumChildShapes();i++)
		{
			compound2->addChildShape(compoundShape->getChildTransform(i)*principal.inverse(),compoundShape->getChildShape(i));
		}
#endif
		delete compoundShape;
		createRigidBody(1.0 , transform, compound2);
	}
	`

srajangarg, 2017-10-02:
Thanks, will test these changes out. Will you accept a PR adding a compound shape example?

On a side note, I notice V-HACD does not have a `CMakeLists.txt` making it impossible to use vhacd if someone is building using cmake, out of the box. I can push in a PR for that too.

erwincoumans, 2017-10-02:
Here is a git link, github messes up the code snippet:
https://gist.github.com/erwincoumans/b8a110b5c228b378161068cd25fa7d2c

Sure, a minimal example, based on BasicExample.cpp would work, same for V-HACD cmake. 
Thanks!

erwincoumans, 2017-10-02:
Note that version0 will create a compoundShape inside compound2, which is less efficient, the other version will re-create a new compound2 with the child box shapes in there, applying the inertial shift.

erwincoumans, 2017-10-02:
For V-HACD there was some PR about it here: https://github.com/bulletphysics/bullet3/pull/477 Would probably be good to recover some example/demo again with V-HACD

closed 2017-10-03 by srajangarg