bulletphysics/bullet3 #1354: Not able to Run PyBullet c# binding inside Unity

balajiraghuram92, 2017-10-03:
Hi,

Not able to Run PyBullet c# binding inside Unity.

Always getting DllNotFoundException "pybullet_vs2010_x64_release.dll" not found.

Even after placing the Dll in the Plugins Folder inside Unity.

When Changing in Native.cs from "pybullet_vs2010_x64_release.dll" to "pybullet_vs2010_x64_release" getting failed to load "pybullet_vs2010_x64_release.dll" with error the Specified Module could not be found.

balajiraghuram92, 2017-10-03:
Fixed the Issue by placing the Python36.dll into plugins folder

closed 2017-10-03 by balajiraghuram92

erwincoumans, 2017-10-03:
Thanks for the report. Note that those bindings are a starting point: we need some way to create companion Unity graphics objects (when loading a URDF from pybullet) and create a pybullet collder/multibody when using the Unity GUI to create a rigid body. In addition, some 'convenience' layer in C# so it behaves similar to pybullet Python would make it a bit easier to use.

balajiraghuram92, 2017-10-04:
Hi,

Loading of URDF needs a path or some address,because when i try to load always get "cmd_URDF_Load_Failed".

Replicating the Pybullet in Unity is Hard as we have to Just Give the Physics support only ,all the rendering and Loading of Object are maintaned by itself.

So we got to find some way sending the Graphics data to Pybullet and Getting URDF data & physics from PyBullet.

Searching in the Bullet Physics Code base to figure out how can we achive this.

Thanks for the Extention of PyBullet for Unity.

erwincoumans, 2017-10-04:
Copy the data folder next to the plugin (pybullet_vs2010_x64_release.dll) so the URDF file can be loaded. Alternatively, use the b3SetAdditionalSearchPath to point to the Bullet/data folder.

We can use Bullet Unity plugin to figure out how to create Unity graphics objects and how to synchronize the transforms.

erwincoumans, 2017-10-04:
Here is the updated script that also gets the position/orientation of a cube, sets gravity etc:
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/unity3d/examples/NewBehaviourScript.cs
Video of the script running in Unity, with external ExampleBrowser as 'debug visualizer':
https://www.youtube.com/watch?v=81dthrVI2vs

balajiraghuram92, 2017-10-05:
Thanks for Helping out.

Now it is working.

As you shown in the video do we have to run the Physics server in the Example browser.

No way just sending details to the Server and Directly Getting Physics Data.

Because if the Movement of the Object must be made in Unity using mouse or VR.

How can we Achive That Using the PyBullet?

Again Thank You Very Much for your Contribution on Plug-in for Unity in PyBullet.