bulletphysics/bullet3 #1369: textOrientation surprising behavior in VR

jonathantompson, 2017-10-08:
If textOrientation is not set when calling addUserDebugText, the debug text has a black background and is scaled appropriately.

When textOrientation is specified, the text elements have space between them (each individual char has a black border). I would expect behavior to be somewhat consistent.

jonathantompson, 2017-10-08:
I'll try and get around to looking at the code and submit a PR if I have time this week.

erwincoumans, 2017-10-10:
It is likely some OpenGL blend state issue (glBlend enable and/orfunction etc)

jonathantompson, 2017-10-11:
I'm not doing anything strange on the OpenGL side? Is there any debug information you need from me?

erwincoumans, 2017-10-11:
It is going to take a bit of time to track down, I'm not asking anything from you, just mentioning in this issue what I think is causing this. The OpenVR OpenGL code is from https://github.com/ValveSoftware/openvr/blob/master/samples/hellovr_opengl/hellovr_opengl_main.cpp, but the OpenGL init is from mine.

erwincoumans, 2017-10-11:
I fixed the black border thing, it was an obvious ordering issue (transparent text needs to be rendered after opaque objects). Note there are a couple of text options that affect how text is rendered, such as camera facing orientation and truetype fonts (by default bitmap fonts are used). I'll expose some option flags and look into the issue of providing orientation.

erwincoumans, 2017-10-11:
Border is fixed. Also added documentation in pybullet quickstart guide about difference in rendering of camera facing (without textOrientation) using bitmap fonts, and text with orientation using TrueType fonts. https://github.com/bulletphysics/bullet3/pull/1378

closed 2017-10-11 by erwincoumans

jonathantompson, 2017-10-11:
Thanks Erwin!