bulletphysics/bullet3 #1375: collision detection: btConeShape behaves like cylinder shape.

EastskyKang, 2017-10-09:
I am using bullet collision for collision detection. 

As I define cone object as follows:

  btCollisionShape* coneCollisionShape = new btConeShapeZ(radius, height);
  btCollisionObject* coneCollisionObject = new btCollisionObject();
  coneCollisionObject->setCollisionShape(coneCollisionShape);

object occupies cylindrical space, not cone shape. Am I missing something here?

erwincoumans, 2017-10-09:
The RollingFrictionDemo uses cone shapes and they work fine.
https://github.com/bulletphysics/bullet3/blob/master/examples/RollingFrictionDemo/RollingFrictionDemo.cpp
Can you create a btRigidBody instead of btCollisionObject?

erwincoumans, 2017-10-09:
How do you determine what space is occupied?

EastskyKang, 2017-10-09:
Thank you so much for your response. 

As I put an unit sphere to certain position (which is not supposed to be occupied), collision was detected. 

By the way, I only want to use collision detection feature, since I have my own dynamics engine. Is using btRigidBody proper way to do it?

erwincoumans, 2017-10-09:
Why using your own dynamics engine, since it obviously works well with btRigidBody, LOL?
Now seriously, it should work just fine. How do you perform the query? Do you set the world transforms for the query properly? What is the contact point information (closest distance, contact normal etc)? Note that a positive distance means no collision.

EastskyKang, 2017-10-10:
@erwincoumans Thank you for comment. I am actually building a dynamics simulator for robotics (for some special purposes). That's why I am using bullet for just for collision detection. 

And I realized I was detecting collision by using collision between manifolds. As I changed my implementation it seems works fine. 

Anyway, thank you so much for your quick and kind response. I will close the issue.

closed 2017-10-10 by EastskyKang

erwincoumans, 2017-10-10:
Robotics is the main focus of Bullet nowadays, please check http://pybullet.org There is URDF  and SDF loading, Inverse Dynamics, Inverse Kinematics, motor control etc. We simulate KUKA arm grasping, quadruped/biped robot locomotion etc.

erwincoumans, 2017-10-10:
See some new paper using pybullet:https://stanfordvl.github.io/ntp/?utm_content=buffer8b1fc
Also our researchers at Google Brain and X use this simulator in both C++ and Python.

EastskyKang, 2017-10-10:
@erwincoumans Great! thank you so much for your suggestion. I will take a look and consider use bullet dynamics engines.