bulletphysics/bullet3 #1379: Inverse Dynamics has problems with fixed joints

MichaelLutter, 2017-10-13:
I tried simple inverse dynamics control with two two-joint-robots. The first robot contains fixed joints between both revolute joints while the second does not contain fixed joints between the revolute joints. 

- For the robot **without fixed joints**, inverse dynamics control **works fine**. The joint position & velocities behave as the desired positions and velocities. 
[image]

- For the robot **with fixed joints** only the first/base joint yields the desired behavior. The second joint **does not track the desired position/velocity**. If the second joint is actuated by a constant the coupling of the first/base joint is not compensated and hence does not behave as expected. 
[image]

For recreation purposes I attached both urdf files as well as the python script for a simple inverse dynamics control example. The figures were created using Pybullet 1.5.6 installed via pip. 

[InvDynamicsTest.tar.gz](https://github.com/bulletphysics/bullet3/files/1382099/InvDynamicsTest.tar.gz)

erwincoumans, 2017-10-13:
Thanks for the reproduction case, it must be some indexing problem between Python and the inverse kinematics (which handles fixed joints just fine). I'll look into it right now.
Once I fixed it, can we add your reproduction case as an example for pybullet?

MichaelLutter, 2017-10-13:
Thanks for the fast response. 

Yes, you can add the my code as pybullet example. 

Side Remark: 
Can you make the disabling of the motors for torque control in the [Quick Start Guide page 17](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#) more explicit? It took me quite a while to figure out what it really meant.

erwincoumans, 2017-10-13:
It was indeed a mismatch, the inverse dynamics excepts input arrays size of #dofs, but pybullet checks for #joints. It is fixed in this CL: https://github.com/bulletphysics/bullet3/pull/1381
Please run "pip install -U pybullet" to update to pybullet version 1.5.8 for the fix.
The updated script is here:
https://github.com/erwincoumans/bullet3/blob/master/examples/pybullet/examples/inverse_dynamics.py#L79
Thanks for the report!

closed 2017-10-13 by erwincoumans

erwincoumans, 2017-10-14:
While looking into this, we are currently using DLS instead of SDLS. SDLS has better behaviour in case the target is out-of-reach (DLS oscillates, SDLS doesn't).