bulletphysics/bullet3 #1380: Return Value of Calculate Inverse Kinematics

MichaelLutter, 2017-10-13:
The _calculateInverseKinematics_ function does not return the correct joint configuration to the corresponding point in task space. _calculateInverseKinematics_ only returns a **joint configuration that is closer to specified point in task space than the current joint configuration**. Therefore, the return values of the inverse kinematics are dependent on the current joint configuration. This configuration dependency was already mentioned in issue #1249. Only if the current joint configuration is close to the IK-solution, the correct joint configuration is returned.

**Is the behavior desired or not?** 
If this behavior is desired, could you please specify the behavior in the pybullet quick start guide, as this is not the behavior I expected. 

This state dependency can be observed in the [inverse kinematic example](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/inverse_kinematics.py). As shown below the Task-space controller can follow the cirlce in task-space. However, n-iterations, which is equivalent to calling the _calculateInverseKinematics_ function n-times, are required to compute the correct desired joint configuration. 

[image]

More closely, the IK computation needs around 5 iterations to converge to the correct joint configuration. As this example does not simulate the joints but sets the joints (via _resetJointState_) to the IK solution in every iteration, I would expect the convergence to the correct configuration from the first time-step.   

[image]

The plots were generated using the [inverse kinematic example](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/inverse_kinematics.py) with the following parameters:

rp = [0,0,0,0,0, 0,0]
useOrientation = 0
useSimulation = 0
useRealTimeSimulation = 0

erwincoumans, 2017-10-13:
Thanks for the reproduction case, it must be some indexing problem between Python and the inverse kinematics (which handles fixed joints just fine). I'll look into it right now.
Once I fixed it, can we add your reproduction case as an example for pybullet?

erwincoumans, 2017-10-13:
We are using the Damped Least Squares IK method and the samping is set too high.
Try to replace jd=[0.1,0.1,0.1,0.1,0.1,0.1,0.1] by smaller values, such as 0.01 for example.

MichaelLutter, 2017-10-13:
Thanks for the fast answers.

I retried the example with different values for joint damping. For me the smaller joint damping removes the smoothing but does solve the problem of using IK as one-shot-predictor. Especially if the current joint configuration is far from the desired joint configuration.  

Joint Damping: 1e-1
[image]

Joint Damping: 1e-3
[image]

Joint Damping: 1e-6
[image]

erwincoumans, 2017-10-13:
I suppose you meant "but does NOT solve the problem of using IK as one-shot-predictor"?
We add an iterative method with a tolerance and some iterations to converge to the exact solution. 
Aside from the damping, the DLS method uses the Jacobian of the current pose, and the further away from the final pose, there will be some issue.
Let me discuss it here with someone who is more familiar with the ins and outs of the IK (Yunfei Bai).

erwincoumans, 2017-10-13:
Indeed, some simple iterative scheme in pybullet, using a tolerance, could work very well. We likely don't need many iterations in general. 

Our underlying inverse kinematics library has various algorithms (ranging from selectively damped least squares (SDLS), DLS, based on Samuel Buss IK library with many improvements: we added the null space method, compute the Jacobian using our Inverse Dynamics library, added orientation targets, velocity limits etc.

This is according to the SDLS paper at the top of page 3, http://math.ucsd.edu/~sbuss/ResearchWeb/ikmethods/SdlsPaper.pdf

"The update (2) to the angles may be done once per frame so that the end effectors only approximately follow the target positions, or it can done iteratively until the end effectors are sufficiently close to the targets."

Currently we only do the update once per call to calculateInverseKinematics.

MichaelLutter, 2017-10-16:
Yes, I meant it does NOT solve the problem of using IK as an one-shot-predictor. 

I think using an iterative IK would be a good idea. The user could specify an error tolerance and an maximum number of iterations. Therefore, the user could trade-off estimation error and computation time.

erwincoumans, 2017-10-25:
Here  is a small python script that does this: https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/inverse_kinematics_husky_kuka.py#L71

`def accurateCalculateInverseKinematics( kukaId, endEffectorId, targetPos, threshold, maxIter):
	closeEnough = False
	iter = 0
	dist2 = 1e30
	while (not closeEnough and iter<maxIter):
		jointPoses = p.calculateInverseKinematics(kukaId,kukaEndEffectorIndex,targetPos)
		for i in range (numJoints):
			p.resetJointState(kukaId,i,jointPoses[i])
		ls = p.getLinkState(kukaId,kukaEndEffectorIndex)	
		newPos = ls[4]
		diff = [targetPos[0]-newPos[0],targetPos[1]-newPos[1],targetPos[2]-newPos[2]]
		dist2 = (diff[0]*diff[0] + diff[1]*diff[1] + diff[2]*diff[2])
		closeEnough = (dist2 < threshold)
		iter=iter+1
	#print ("Num iter: "+str(iter) + "threshold: "+str(dist2))
	return jointPoses`

And a video of that IK: https://www.youtube.com/watch?v=qZiR6vNcmCU

MichaelLutter, 2017-11-01:
Yes, this is a possible solution. But that does not solve my problem as I cannot call _resetJointState_ during a simulation. 

I looked at the Inverse Kinematics implementation and to completely solve the problem one would have to expose the starting joint configuration in the Python API and a Forward Kinematics function that computes task-position from a joint configuration without changing the simulation state. I personally don't know how much effort this forward kinematics function would be and how it would be realised.

jonathantompson, 2018-05-24:
Ping on this as well. It would be great if we could expose the seed joint configuration from C++. By default it does make sense to use the current joint configuration, but there are other cases where using a rest pose for the IK method makes sense as well.

I'm also finding that the settling behavior of IK is adding latency. Decreasing the dampening can result in instability, but is also less than ideal (I'd rather find a solution, if it exists without having to step the simulator or hard-set joints).

It would also be good to return / expose a few things:

1. The final residual of the solver and / or the solver's best fit endeffector pose at termination.
2. The number of solver steps to take.
3. Solver residual termination condition.

Thoughts?

erwincoumans, 2018-05-24:
>> It would also be good to return / expose a few things:

Those things are all implemented and exposed in the accurateCalculateInverseKinematics script. Are you suggesting implementing accurateCalculateInverseKinematics in C++?

A few more suggestions: PyBullet can connect to multiple independent simulations in the same thread, and you can duplicate the robot in a 'scratch' simulation and do your IK using 'accurateCalculateInverseKinematics'.

pseudo scenario:

import pybullet as p #only used for constants
from pybullet_utils import bullet_client

mainSim = bullet_client.connect(p.GUI)
ikSim = bullet_client.connect(p.DIRECT)

#duplicate the robot from mainSim to ikSim, either manually or using pybullet_utils.UrdfEditor
#do your ik in ikSim, using ikSim.resetJointState and ikSim.calculateInverseKinematics APIs.

mainSim.stepSimulation()

You can also use pybullet.saveState and pybullet.restoreState to temporarily make changes and revert them.

erwincoumans, 2018-05-24:
Yes I agree, we should add the option to allow the user to provide the initial IK pose (similar to what we allow for calculateInverseDynamics). Let me discuss with Yunfei, shouldn't be too hard for us to add.

YunfeiBai, 2018-05-24:
Sounds good, we can discuss it when we are back next week.

erwincoumans, 2018-05-31:
I finally implemented the accurate inverse kinematics in C++, and exposed through PyBullet.calculateInverseKinematics. Now the default  residual threshold is 1e-4 with 20 max iterations. Pull request was merged: https://github.com/bulletphysics/bullet3/pull/1722
Available in PyBullet 2.0.1, pip install pybullet --no-cache-dir
Tested with this Baxter robot demo from Tom Silver:
https://github.com/erwincoumans/pybullet_robots

closed 2018-05-31 by erwincoumans