bulletphysics/bullet3 #1391: Dynamic linking in MSVC

Chainsawkitten, 2017-10-19:
Dynamic linking options are provided via the CMake option `BUILD_SHARED_LIBS`. However, simply stating that a library should be compiled as SHARED isn't enough in MSVC. In MSVC, you need to use `__declspec(export)` when compiling the library and `__declspec(import)` when using it, in order to export the functions/methods that should be visible to users of the DLL.

As far as I can see, this isn't currently done (I've looked at LinearMath). This means the library can't be used as a DLL when compiling with MSVC since no import library files are written.

The solution would be to define a macro:
```
#ifdef _WIN32
#ifdef LinearMath_EXPORTS
#define LINEARMATH_API __declspec(dllexport)
#else
#define LINEARMATH_API __declspec(dllimport)
#endif
#else
#define LINEARMATH_API
#endif
```
and use this macro when declaring a function/method that should be accessible to the user (so practically everywhere).

(`LinearMath_EXPORTS` is a definition CMake adds when building a shared library on Windows.)

erwincoumans, 2017-10-19:
We have the C-API that uses dllexport on its API, see here:
https://github.com/bulletphysics/bullet3/blob/master/examples/SharedMemory/PhysicsClientC_API.h
There are some C# bindings that use this C-API, and it seems work just fine in Unity, even without dllimport. Please have a look at it.

Chainsawkitten, 2017-10-19:
Cool. Is there any particular reason why the C++ API doesn't do it? Or is it just that it's a lot of work no one has found worth doing?

Harha, 2019-01-07:
I'm currently unable to build the CMake generated MSVC 2017 projects if I choose BUILD_SHARED_LIBS. All the bullet related projects fail due to linker errors (.lib file missing?? says MSVC).

I tried a fresh copy of the latest master branch (c993175a5859e48783cf2c490947d826be5649b2) and then release 2.88. Both fail in the same manner.

I'm personally just looking for a way to link the library with my game project written in C, so that I'd be able to command bullet via the C-api (PhysicsClientC_API.h).

I tried generating the MSVC project files using default CMake settings at first, this built bullet3 just fine with no errors but in a static manner. Then I tried linking the .lib files with my C game project and calling functions using the C-api header file. b3ConnectSharedMemory and b3DisconnectSharedMemory seem to initially work fine (the first one returns a pointer to the client handle), but if I try to call b3InitResetSimulationCommand (or any other b3 function that takes the client handle as an argument), I get an access violation exception (write at some address).

I'm not personally sure if the C-api for Bullet3 is event intended to work when linked statically? And what would be the correct way to build bullet3 dynamically?

Harha, 2019-01-08:
I'm not sure how off-topic this problem of mine is, but can @erwincoumans shed any light into this?

If you need more information I can provide. I've tried even converting my whole game project to C++ to be compiled as C++ code under visual studio, still the same exact problem. I've tried isolating these bullet3 function calls to the start of my main function before anything else, same problem. I've tried different linker flags, still same problem. I'm out of ideas and have very little experience about linking a C program against a C++ library with a C api anyways.

[image]

erwincoumans, 2019-01-08:
Indeed, for C++ we don't expose symbols. You should be able to statically link against Bullet without issues, no need for dynamic DLL linking.

The C-API should be able to link dynamically / DLL, since it is used in a Python extension plugin.

If you use b3ConnectSharedMemory, that implies you connect to an existing running physics server.
You likely want to b3ConnectDirect (no debug GUI) or b3ConnectGUI(...).
After you connect, make sure you can send commands.

closed 2019-01-08 by erwincoumans