bulletphysics/bullet3 #1397: Adding friction for concave shape

hzyjerry, 2017-10-22:
Hi, I have a 3D building model that I would like to import into pybullet as "environment", it is stored in obj format. The documentation suggests I use `createCollisionShape` and `createMultiBody` to import the raw obj file with `GEOM_FORCE_CONCAVE_TRIMESH` flag. 

This method works, however, I would also like to add friction to the surface of the model, which I don't think is supported for raw obj file yet (is it?). I tried wrapping the obj file inside an urdf file with explicit friction, however the imported shape is not "concave", as in it doesn't create a "concave building environment".

Thus I would like to know what's the best way to add friction for a concave obj shape? Thanks!

hzyjerry, 2017-10-23:
Figured it out, you can explicit set these values with `p.changeDynamics(boundaryUid, -1, lateralFriction=0.8, spinningFriction=0.1, rollingFriction=0.1)`, where `boundaryUid` is the collision shape ID.

Works great!

closed 2017-10-23 by hzyjerry

erwincoumans, 2017-10-23:
Yes, that's right. You can also import concave shapes in a URDF file, using the non-standard concave="yes" attribute in the link:
 <link concave="yes" name="baseLink">
See also https://github.com/bulletphysics/bullet3/blob/master/data/samurai.urdf