bulletphysics/bullet3 #1400: Raycast vehicle shaking when resting

hiker, 2017-10-23:
In supertuxkart we are using  raycast vehicles from bullet, and I noticed that karts would shake when standing still. Karts would have a linear velocity of around +-0.2 to +-0.3 m/s forward/backward in different frames when no force was applied and the kart were basically at a standstill.

This seems to be caused by the friction handling of the raycast vehicle: calcRollingFriction would return just positive or just negative values.
While I didn't study the physics behind that part of the code, I guessed that because of the comment:
```
        // calculate j that moves us to zero relative velocity
```
it does calculate one impulse to be applied  to slow down the kart's rolling - but is ignoring the fact that there are actually 4 (or whatever) wheels - resulting in a 4-times too large impulse to be applied, causing the kart to reverse direction. So I divided the j1 value returned by calcRollingFriction by the number of wheels, and since then the shaking is basically gone. 
I can't say if this is indeed a bug in the physics, or if I was just lucky that the changed parameter worked in our case :) But I think it's worth having a look.

erwincoumans, 2017-10-24:
It has been a while, good to hear from you.
That sounds good, thanks for the report! Do you mind to create a patch for this?

erwincoumans, 2017-10-26:
Here is a suggested fix: https://github.com/bulletphysics/bullet3/pull/1407
Please re-open if it doesn't work for you. Thanks!

hiker, 2017-10-28:
Sorry for the delay, we've been working on our next release candidate.. I can't use the patch we applied in SuperTuxKart directly, but here an admittedly untested port of the STK patch:
```
$ git diff ../src/BulletDynamics/Vehicle/btRaycastVehicle.cpp
diff --git a/src/BulletDynamics/Vehicle/btRaycastVehicle.cpp b/src/BulletDynamics/Vehicle/btRaycastVehicle.cpp
index a7b168846..130ff9088 100644
--- a/src/BulletDynamics/Vehicle/btRaycastVehicle.cpp
+++ b/src/BulletDynamics/Vehicle/btRaycastVehicle.cpp
@@ -493,8 +493,8 @@ struct btWheelContactPoint

 };

-btScalar calcRollingFriction(btWheelContactPoint& contactPoint);
-btScalar calcRollingFriction(btWheelContactPoint& contactPoint)
+btScalar calcRollingFriction(btWheelContactPoint& contactPoint, int numWheelsOnGround);
+btScalar calcRollingFriction(btWheelContactPoint& contactPoint, int numWheelsOnGround)
 {

        btScalar j1=0.f;
@@ -513,7 +513,7 @@ btScalar calcRollingFriction(btWheelContactPoint& contactPoint)
        btScalar vrel = contactPoint.m_frictionDirectionWorld.dot(vel);

        // calculate j that moves us to zero relative velocity
-   j1 = -vrel * contactPoint.m_jacDiagABInv;
+ j1 = -vrel * contactPoint.m_jacDiagABInv/numWheelsOnGround;
        btSetMin(j1, maxImpulse);
        btSetMax(j1, -maxImpulse);

@@ -615,7 +615,7 @@ void        btRaycastVehicle::updateFriction(btScalar       timeStep)
                                        btScalar defaultRollingFrictionImpulse = 0.f;
                                        btScalar maxImpulse = wheelInfo.m_brake ? wheelInfo.m_brake : defaultRollingFrictionImpulse;
                                        btWheelContactPoint contactPt(m_chassisBody,groundObject,wheelInfo.m_raycastInfo.m_contactPointWS,m_forwardWS[wheel],maxImpulse);
-                                   rollingFriction = calcRollingFriction(contactPt);
+                                 rollingFriction = calcRollingFriction(contactPt, numWheelsOnGround);
                                }
                        }

```
Main differences are

1. that I divide by the number of wheels actually touching the ground (which probably makes hardly any difference in reality)
2. In STK I divide in calcRollingFriction before capping j1 - and I admit I am not sure what is actually correct :) But also I don't think it will make much of a difference (since the kart has to be on close to 0 speed in order for the rolling friction to be noticeable).

So in short: I reopen this so you are aware of the difference and can decide which is correct, I am happy with either patch :)    [though of course the changes to Extras/VHACD/test/src/main.cpp appear to be unrelated].

Thanks for your help!
   Joerg

erwincoumans, 2017-10-29:
Thanks, I can amend it. Where are those changes been made? I didn't see it in https://github.com/supertuxkart/stk-code/blob/master/lib/bullet/src/BulletDynamics/Vehicle/btRaycastVehicle.cpp#L485

Also, did you ever work on power sliding in corners? At some stage a nicer vehicle model would be great.

hiker, 2017-10-29:
Hi,

sorry, we created our own copy of the raycast vehicle, so we could keep an as-clean-as-possible copy of bullet in case that we want to update (which in the end we never did). 
https://github.com/supertuxkart/stk-code/commit/4c1874ebf86ab6cd85dc92846cda0e83fa7a2fba

Re skidding: we have that implemented, but we are cheating: when you skid, we tweak the turn radius, add a graphical only rotation (with separate raycast to find location for skidmarks). Then on release of the skid we align graphical and physical view again, and add a speed boost. So no real skidding

Re "nicer vehicle model": do you mean the graphical models? There have been quite a few updates in our graphics (though not all karts have been updated). Or the physical model?

I admit we still have some strange physics behaviour - sometimes karts can drive up on walls, and sometimes karts would get pushed backed far too much (the latter caused by high speed collision: kart and track intersect too much --> the computed impulse to resolve the collision is too big, and often contains an 'up' component (likely caused by slight differences in suspension I guess), which then sends the kart flying up into the air. We have actually halved the physics timestep size, and that got rid of the most severe issues).

erwincoumans, 2017-12-01:
Thanks for the feedback, I modified it to divide by the number of wheels, as you suggested, so let's close the issue. If you have some other (easy to reproduce) issue with Bullet, please report it here again.

closed 2017-12-01 by erwincoumans