bulletphysics/bullet3 #1401: Bullet3 vs Bullet2

GregSlazinski, 2017-10-24:
I can't seem to find any information about Bullet2 vs Bullet3.
The repository is called Bullet3, but it has both 2 and 3 versions inside?
Which one should I be using?
Are they both stable and ready for production use?
What's the API difference? (small differences or complete API rewrite)
When is Bullet3 going to be released?
Thanks

erwincoumans, 2017-10-24:
You just use the Bullet 2.x part, which is in Bullet/src/BulletDynamics, Bullet/src/BulletCollision etc. 
The Bullet3OpenCL and related directories can be ignored, there is no progress on that, and no current plan, since we focus on pybullet nowadays, which is a layer on top of Bullet 2.x and some example infrastructure, adding inverse kinematics, inverse dynamics, VR etc. See http://pybullet.org

closed 2017-10-24 by erwincoumans

mode89, 2022-02-23:
Hi @erwincoumans :)

Are there any changes to this? Is it still preferable to use Bullet 2.x?