bulletphysics/bullet3 #1402: IK solution is problematic after setting the orientation for the robot

kuanfang, 2017-10-24:
I found that setting the orientation for the robot can cause problems for IK. Unless the orientation is set to [0, 0, 0, 1], the IK solution could be wrong. With larger deviation from [0, 0, 0, 1], the IK solution would be more problematic.

The problem can be demonstrated by the code below (please make sure you have the `kuka_iiwa/model.urdf` from the data folder):

`
#!/usr/bin/env python
import argparse

import numpy as np

import pybullet

def parse_args():
    parser = argparse.ArgumentParser()

    parser.add_argument(
            '--debug',
            dest='debug',
            type=int,
            default=0,
            help='Set to 1 to see the working version.')

    args = parser.parse_args()

    return args

def main():
    args = parse_args()

    client_uid = pybullet.connect(pybullet.GUI)
    pybullet.resetSimulation()
    pybullet.setGravity(0,0,0)
    pybullet.setRealTimeSimulation(0)
    pybullet.setTimeStep(1e-3)

    LIMB_URDF = ('kuka_iiwa/model.urdf')

    if args.debug:
        base_quat=[0, 0, 0, 1]
    else:
        base_quat=[0, 0, 0.707, 0.707]

    sawyer_uid = pybullet.loadURDF(
            LIMB_URDF,
            basePosition=[0, 0, 0],
            baseOrientation=base_quat,
            useFixedBase=True,
            )
    ee_link_ind = 6

    num_joints = pybullet.getNumJoints(sawyer_uid)
    for joint_ind in range(num_joints):
        pybullet.resetJointState(sawyer_uid, joint_ind, 0)

    target_pos = [0, 0, 0.7]

    t = 0
    while(1):
        joint_positions = pybullet.calculateInverseKinematics(
                bodyUniqueId=sawyer_uid,
                endEffectorLinkIndex=ee_link_ind,
                targetPosition=target_pos,
                )

        for joint_ind, joint_position in enumerate(joint_positions):
            pybullet.resetJointState(
                    bodyUniqueId=sawyer_uid,
                    jointIndex=joint_ind,
                    targetValue=joint_position)

        if t == 1000:
            pos, quat, _, _, _, _ = pybullet.getLinkState(
                    sawyer_uid,
                    ee_link_ind)

            print(pos, quat)

        pybullet.stepSimulation()
        t += 1

if __name__ == '__main__':
    main()
`

If you set the orientation to [0, 0, 0, 1] by running `python experiments/debug_ik_2.py --debug 1`, the result will make sense. 

If you set the orientation to [0, 0, 0.707, 0.707] by running `python experiments/debug_ik_2.py --debug 0`, the result will make sense.

erwincoumans, 2017-10-24:
The end-effector location/orientation needs to be computed in the base frame of the robot, not in world space. Are you doing this? (there is some helper function to compute the inverse etc)

kuanfang, 2017-10-24:
But in the [quickstart/calculateInverseKinematics](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.9i02ojf4k3ve) , the instruction said the position and the orientation should be in "Cartesian world space". Is it outdated?

erwincoumans, 2017-10-24:
You are right, the end effector target position and orientation are in Cartesian world space coordinates. Let us have a look what is happening and get back to you.

erwincoumans, 2017-10-24:
This pull request should deal with it:https://github.com/bulletphysics/bullet3/pull/1404
We have only used the IK for fixed robots with identity transform for the base. This should allow for arbitraty transforms and moving base etc.

erwincoumans, 2017-10-25:
Fixed, with a video specially for you :-)
https://www.youtube.com/watch?v=qZiR6vNcmCU
and a test: https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/inverse_kinematics_husky_kuka.py
requires pybullet 1.6.1 (pip install -U pybullet)

closed 2017-10-25 by erwincoumans

kuanfang, 2017-10-25:
Great! Thank for the prompt response Erwin!

animesh-garg, 2017-10-25:
Thanks a lot Erwin