bulletphysics/bullet3 #1403: ModuleNotFoundError: No module named 'pybullet_envs.bullet.cartpole_bullet'

jgresham, 2017-10-24:
**Steps to reproduce**
Fresh install of pybullet with python3.6, then run
`python3 -m pybullet_envs.baselines.train_kuka_grasping`

Removing
`from pybullet_envs.bullet.cartpole_bullet import CartPoleBulletEnv`
in /pybullet_envs/baselines/enjoy_pybullet_cartpole.py and /pybullet_envs/baselines/train_pybullet_cartpole.py fixed the issue.

benelot, 2017-10-24:
Thanks for the input, it is great that somebody starts to use the envs! I
will check if I can reproduce that problem. Weird is that you get problems
during kuka training with the cartpole training/running scripts.

erwincoumans, 2017-10-26:
importing some of the modules (pybullet_envs etc) will import various packages. It should be fixed in latest release (this commit: https://github.com/bulletphysics/bullet3/pull/1405/commits/9385f36505a37f99e6d2c92bd2c052d25305ea2a) I'll update the pypi pybullet package now.

closed 2017-10-26 by erwincoumans

jgresham, 2017-10-28:
Thank you for the prompt reply and fix.

erwincoumans, 2017-10-28:
The train_kuka_grasping and some other environments are still work-in-progress by the way, we'll make them train using DQN (discrete) and PPO (continuous), same for racecar and minitaur.