bulletphysics/bullet3 #1411: Unable to install on Win10: pip install pybullet

bingjeff, 2017-10-29:
Python 2.7 installed using Anaconda 5.0.

Also unable to install pybullet using the python setup.py install unless I comment out the numpy option.

Is it possible that there is just a mismatch on the numpy version? The compiler was very unhappy with the numpy specific c-types.

Everything works just fine if I install with numpy flag removed.

erwincoumans, 2017-10-29:
Thanks for the report. Did you try to install numpy first (pip install numpy)?

bingjeff, 2017-10-29:
Yeah. Numpy was installed. Presumably the wrong version. There were a bunch
of warnings about using a deprecated numpy API.

I'm not super familiar with this, but if it's helpful I can try and figure
out how to compile on my system.

Alternatively, if I know the approved method of installation I can do that.

erwincoumans, 2017-11-11:
No, it was a C99 issue, variables need to be declared at the beginning of a block. Fixed now, please update.
Thanks for the report!

closed 2017-11-11 by erwincoumans

bingjeff, 2017-11-11:
Awesome! Thanks Erwin.