bulletphysics/bullet3 #1414: Andoid compilation error - Godot

AndreaCatania, 2017-10-31:
Related Godot issue: https://github.com/godotengine/godot/issues/12347

A user report that is not possible compile Godot with Bullet for Android.

By looking the log the problem seems here: https://github.com/bulletphysics/bullet3/blob/master/src/LinearMath/btAlignedObjectArray.h#L39

```
thirdparty/Bullet/src/LinearMath/btAlignedObjectArray.h:39:10: fatal error:
'new' file not found
#include //for placement new
^~~~~
1 error generated.
scons: *** [thirdparty/Bullet/src/BulletCollision/BroadphaseCollision/btAxisSweep3.android.opt.debug.armv7.neon.o] Error 1
scons: building terminated because of errors.
````

Check here for more info: https://github.com/godotengine/godot/issues/12347

I saw this comment https://github.com/bulletphysics/bullet3/blob/master/src/LinearMath/btAlignedObjectArray.h#L23-L27

Is possible to add a flag that allow me to change it during compile time depending on compilation target? (Godot side I'll change this flag with scons)

erwincoumans, 2017-10-31:
Yes, the macro is useless right now anyway.
How about introducing 
#ifndef BT_AVOID_PLACEMENT_NEW
#include <new>
#else 
#do whatever android needs
#endif //BT_AVOID_PLACEMENT_NEW
A suggested patch is welcome.

AndreaCatania, 2017-11-03:
@erwincoumans Thanks for support. I resolved by enabling STL on Android compiler. Nothing is needed in bullet side.

closed 2017-11-03 by AndreaCatania