bulletphysics/bullet3 #1416: Performance Issues w.r.t. Computation Time

MichaelLutter, 2017-11-01:
I am experiencing quite severe performance issues w.r.t. to computation time. When I actually scale to better hardware the computation time per simulation step increases by an order of magnitude, even though more resources are being used. An 3 year old i5 outperforms a new  i7 by one order of magnitude, even tough the i7 devotes three cores to the bullet computations.

What could be causing such behaviour? 
How could such an behaviour be resolved? 
What is the fastest connection for PyBullet? Shared Memory or Direct?

Furthermore, the shared memory physics server uses two complete cores even when no simulator is connected. How could the resource usage be decreased if no simulation is executed? 

### Performance Benchmark:
#### Total Computation Time:
| Script  |  1) | 2)  | 3) |
| ------------- | ------------- | ------------- | ------------- |
| MultiBody Sim (Direct Gui) | -  | -  | 57.457s  |
| MultiBody Sim (Shared Memory)  | 136.078s | 10.463s | 29.670s | 
| MultiBody Sim (Direct) |46.093s|6.120s|7.929s  | 
| InvDyn Sim (Direct Gui) | - | - | 1.605s |
| InvDyn Sim (Direct) | 0.940s | 0.102s | 0.040s |
| InvDyn Sim (Shared Memory) | 2.126s |  0.091s | 0.114s |

#### Computation Time per Timestep
| Script  |  1) | 2)  | 3) |
| ------------- | ------------- | ------------- | ------------- |
| MultiBody Sim (Direct Gui) | - | -| 5.742e-02s |
| MultiBody Sim (Shared Memory)  | 1.361e-01s | 1.046e-02s | 2.963e-02s |
| MultiBody Sim (Direct) | 4.609e-02s | 6.141e-03s | 7.928e-03s |
| InvDyn Sim (Direct Gui) | - | - | 8.023e-04s |
| InvDyn Sim (Direct) | 4.698e-04s | 7.025e-05s | 2.021e-05s |
| InvDyn Sim (Shared Memory) | 1.063e-03s | 4.563e-05s | 5.695e-05s |

1) **Ubuntu 16.04:**
- CPU: 		i7-7800X @ 3.50 - 4.00 GHz, 6 Cores, 12 Threads
- Memory: 	32 GB RAM
- Graphics: 	GTX 1070
- Software: 	Python 2.7.12 (system installation) / Pybullet 1.6.2 (Compiled from Source), NVIDA OpenCL & Cuda

2) **Ubuntu 16.04:**
- CPU: 		i5-7400, @ 3.00 - 3.50 GHz, 4 Cores, 4 Threads
- Memory:	8 GB RAM (2133 MHz)
- Graphics: 	GTX 1050
- Software: 	Python 2.7.12 (system installation) / Pybullet 1.6.2 (Compiled from Source)

3) **MacBook Pro 2014 Model**
- CPU: 		i5-4308U @ 2.8 - 3.3 GHz, 2 Cores , 4 Threads 
- Memory: 	16 GB RAM (1600 MHz)
- Graphics: 	Intel Iris 1536 MB
- Software: 	Python 2.7.13 / Pybullet 1.6.2 ((Pip Version))

erwincoumans, 2017-11-01:
Without reproduction cases I can't help.
Can you share the scripts you used to measure performance?

MichaelLutter, 2017-11-01:
I attached the scripts. The scripts are just slight variations of two of the PyBullet examples. The computation time is measured by time.clock. 

 
[PyBulletPerformance.zip](https://github.com/bulletphysics/bullet3/files/1435516/PyBulletPerformance.zip)

erwincoumans, 2017-11-01:
What is script (1), (2) and (3) in your initial post?

pybullet.DIRECT should perform best, it doesn't allocate any new thread etc.

Attached is a modified version of createSphereMultiBodies, with some more info: it creates a performance log in json. You can open this in Google Chrome, using about://tracing and point to that tile. It gives info where performance goes. useMaximalCoordinates will speed things up, and can be useful for static geometry (non-moving, such as samurai) and for non-jointed bodies.

For this example, my timings are around 3ms, as you can see in the image. Most time goes into the constraint solver, since the default number of iterations it 50, very high. You can lower it to 10 or so, to speed things up. Also, apparently pybullet creates compound shapes for spheres, which shouldn't be needed, that can shave off some more time.

Do you get bad timings in DIRECT mode on some platforms?

[image]
[createSphereMultiBodies.zip](https://github.com/bulletphysics/bullet3/files/1435679/createSphereMultiBodies.zip)

erwincoumans, 2017-11-01:
Here is the output for the best case on my Maxbook pro, of the most up-to-date 'createMultiBodies.py' from previous zip file.

python createSphereMultiBodies.py --max=1
pybullet build time: Oct 31 2017 16:18:04
--max=1 (useMaximalCoordinates)
--mode=2(connection mode)
DIRECT
--iterations=10 (constraint solver iterations)
Writing 829011 timings for thread 0

-------------------------------------------------
                 	  Iter  	 Total  
Object Creation  	1.414e-04s	0.018s
Simulation       	1.230e-03s	1.230s
-------------------------------------------------
Simulation       	        	1.230s

Can you run it on some machines using --max=1 and DIRECT mode, and zip the json files and attach them? json name is createSphereAndSimTimings.json_0.json

erwincoumans, 2017-11-01:
with --max=1 and 10 solver iterations, a single simulation step takes around 1.3ms on my machine, which is reasonable:
[image]

MichaelLutter, 2017-11-01:
Thanks for these cool insights. How did you generate the plots? It would be interesting to see how the figures vary between hardware.

Yes, I will try that script on my machines. 

Generally my topic was not about the absolute computation time but the relative computation time between hardware. The 1), 2) & 3) correspond to the hardware configuration which were described below the tables. The macbook pro achieves with an **2014 i5 and one core** a computation time that is **one order of magnitude lower** than the same script running on a **2017 i7 occupying three cores**, where each core should be more powerful than the one core of the i5.

This does not make sense to me. My only guess is that there is some overhead which blocks the computation.

MichaelLutter, 2017-11-01:
Thanks for your extensive help. I ran your create Spheres on all systems using

python createSphereMultiBodies.py --max=1

. The Json Logs can be downloaded from my [Google Drive](https://drive.google.com/file/d/0B390aE8xuBACUkJucjkzMjkwcjQ/view?usp=sharing).  The macbook still achieves the lowest computation time with 1.8 seconds. The computation times are:

1) 8.669s 
2) 2.028s
3) 1.803s

erwincoumans, 2017-11-01:
I'll have a look soon.

OK, shared memory/GUI mode do some threading and sleep, so please ignore them for benchmarks for now.

I assume you used DIRECT for the json files? Please don't use shared memory or GUI, it gets very confusing that way.

So system3=The macbook pro achieves with an 2014 i5 and one core?
System1=2017 i7? What operating system?
What is system2 and which operating system?

Thanks!
Erwin

MichaelLutter, 2017-11-01:
Yes, the JSON's are DIRECT. As DIRECT is the default, I did not explicitly specify it. The Systems are:

#### System 1: 
OS: Ubuntu 16.04:
CPU: i7-7800X @ 3.50 - 4.00 GHz, 6 Cores, 12 Threads
Memory: 32 GB RAM
Graphics: GTX 1070
Software: Python 2.7.12 (system installation) / Pybullet 1.5.8 (Compiled from Source), NVIDA OpenCL & Cuda

#### System 2
Os: Ubuntu 16.04:
CPU: i5-7400, @ 3.00 - 3.50 GHz, 4 Cores, 4 Threads
Memory:	8 GB RAM (2133 MHz)
Graphics: GTX 1050
Software: Python 2.7.12 (system installation) / Pybullet 1.5.8 (Compiled from Source)

#### System 3
OS: 10.12 - MacBook Pro 2014 Model
CPU: i5-4308U @ 2.8 - 3.3 GHz, 2 Cores , 4 Threads
Memory: 16 GB RAM (1600 MHz)
Graphics: Intel Iris 1536 MB
Software: Python 2.7.13 / Pybullet 1.6.2 (Pip Version)

erwincoumans, 2017-11-01:
Thanks for the data. The compiler likely didn't do a good job on Ubuntu.

Could you please use "pip install -U pybullet" on each system? pip will also compile from source anyway.

[image]

erwincoumans, 2017-11-02:
I have some Ubuntu 16.04 laptop here (product: Intel(R) Core(TM) i7-6700HQ CPU @ 2.60GHz)
 and tried to reproduce your issue, but cannot, the timings are 2.8s (default reduced coordinates) and 1.9s (maximal coordinates), see below:

It could be that the Mac OSX compiler (clang etc) just does a better job optimizing than gcc on Ubuntu.

sudo pip install -U pybullet
createSphereMultiBodies.py
-------------------------------------------------
                 	  Iter  	 Total  
Object Creation  	2.570e-04s	0.032s
Simulation       	2.892e-03s	2.892s
-------------------------------------------------
Simulation       	        	2.893s

--------------------------------------------------------------
createSphereMultiBodies.py --max=1
pybullet build time: Nov  2 2017 10:29:24
--max=1 (useMaximalCoordinates)
--mode=2(connection mode)
DIRECT
--iterations=10 (constraint solver iterations)
Writing 831208 timings for thread 0
-------------------------------------------------
                 	  Iter  	 Total  
Object Creation  	1.106e-04s	0.014s
Simulation       	1.996e-03s	1.996s
-------------------------------------------------
Simulation       	        	1.996s

MichaelLutter, 2017-11-02:
So I had time to play around a bit. 
I used the CMake building script with gcc 5.4. 

I tried the pip installation on all systems with gcc 5.4 and gcc 7.2. I did not see a change in performance. The stronger Ubuntu System 1 needs 8.6s to complete while weaker Ubuntu System 2 needs 1.9s. I guess it cannot be an Ubuntu and gcc problem as we could reproduce the 1.9s performance on Ubuntu. For completeness I uploaded the traces, but these should not look to different [Google Drive](https://drive.google.com/file/d/1cVy4u18GiEWv84phaBNaHa5TVvjIRVdI/view?usp=sharing)

OpenCL is turned off by default? So it cannot be caused by GPU computations? As system 1 is the only system with Cuda and OpenCl installed. 

`python createSphereMultiBodies.py --max=1
pybullet build time: Nov  2 2017 19:05:00
--max=1 (useMaximalCoordinates)
--mode=2(connection mode)
DIRECT
--iterations=10 (constraint solver iterations)
Writing 831208 timings for thread 0

-------------------------------------------------
                 	  Iter  	 Total  
Object Creation  	1.988e-04s	0.025s
Simulation       	8.640e-03s	8.640s
-------------------------------------------------
Simulation       	        	8.640s
`

erwincoumans, 2017-11-02:
pybullet only runs on CPU at the moment, please ignore CUDA/OpenCL.

I just did the test on a tiny NVIDIA Jetson TX2 with Ubuntu 16.04 and it runs the benchmark around 5sec on its ARM cpu.

It is not clear why your i7 Ubuntu system runs so slow and I cannot reproduce it on any of my machines. Did you make sure pybullet is removed properly, and re-install the pip version?

MichaelLutter, 2017-11-02:
For the pip installation I used:

```
pip uninstall pybullet
pip install pybullet --user --no-cache-dir
```
From my understanding this should install pybullet from scratch. I also updated the Bios to get the newest Intel Microcode, but the problem still remains. Without Maximal Coordinates the performance degrades even more to 46.5s compared to the 2.8s of the other systems.

Tomorrow, I can try to reproduce the issue on a quite similar CPU. If this does not reproduce the results, I will verify the CPU performance on other benchmarks. 

**Thanks for your super fast & extensive support.** 

```
python createSphereMultiBodies.py 
pybullet build time: Nov  2 2017 23:13:40
--max=0 (useMaximalCoordinates)
--mode=2(connection mode)
DIRECT
--iterations=10 (constraint solver iterations)
Writing 1048576 timings for thread 0

-------------------------------------------------
                 	  Iter  	 Total  
Object Creation  	4.106e-04s	0.051s
Simulation       	4.646e-02s	46.457s
-------------------------------------------------
Simulation       	        	46.457s

python createSphereMultiBodies.py --max=1
pybullet build time: Nov  2 2017 23:13:40
--max=1 (useMaximalCoordinates)
--mode=2(connection mode)
DIRECT
--iterations=10 (constraint solver iterations)
Writing 831208 timings for thread 0

-------------------------------------------------
                 	  Iter  	 Total  
Object Creation  	2.076e-04s	0.026s
Simulation       	8.987e-03s	8.987s
-------------------------------------------------
Simulation       	        	8.987s
```

MichaelLutter, 2017-11-06:
So I investigated this further and found the reason. The delay is caused by __vdso_gettimeofday. So for a currently unknown reason the time "system call"  takes much longer, i.e. System 1: 40s or 86% of 46s vs System 2: 0.037s or 1.28% of 2.9s.

While evaluating the CPU performance using the [Bullet Test of Phoronix Testsuite](https://openbenchmarking.org/test/pts/bullet) I did not experience this time delay. Hence I guess the timing is used within the Pybullet Interface and not the Bullet Core?  
 
These results were obtained using perf on ubuntu 16.04 with
```
perf record -F 99 python createSphereMultiBodies.py
```

**System 1:**
[image]

**System 2:**
[image]

erwincoumans, 2017-11-08:
pybullet adds a lot of performance profiling, which uses _gettimeofday on Linux. I'm very surprised it is so slow. We can figure out a way to make the impact less/allow to disable those timers.

Thanks for tracking it down!

MichaelLutter, 2017-11-09:
So, I finally "resolved" this issue by upgrading my Ubuntu 16.04 LTS to 17.04. Therefore, it seems like an Bug in Ubunutu 16.04, which might be specific to Intel Skylake CPU's as other Intel CPU's on 16.04 were not affected. 

```
-------------------------------------------------
                 	  Iter  	 Total  
Object Creation  	8.229e-05s	0.010s
Simulation       	1.122e-03s	1.122s
-------------------------------------------------
Simulation       	        	1.122s
```

closed 2017-11-09 by MichaelLutter

abishekh, 2018-03-07:
is there a significant performance difference between the python and c++ APIs ? Has some one benchmarked the two to compare ?

dchichkov, 2018-07-25:
I'm getting a nasty amount of syscalls too: 7M calls / second on a EC2 box.   And it looks like these are coming from pybullet.cpython  ::: getTimeMicroseconds.     Is there a way to completely disable profiling/Timers ?

I've adjusted clock source to tsc, and a test shows that normally in a single thread gettimeofday doesn't result in a syscall.  But it looks like from pybullet somehow it does.  On  EC2 it looks like there are issues , even if one had switched the timesource to tsc: https://blog.packagecloud.io/eng/2017/03/08/system-calls-are-much-slower-on-ec2/

```
+   27.81%     5.27%  pybullet.cpython-35m-x86_64-linux-gnu.so    [.] _ZN7b3Clock19getTimeMicrosecondsEv
+   23.09%     9.23%  [vdso]                                      [.] __vdso_gettimeofday
+   15.44%     0.00%  pybullet.cpython-35m-x86_64-linux-gnu.so    [.] _ZN34InProcessPhysicsClientSharedMemoryD0Ev
+   15.44%     0.00%  [unknown]                                   [.] 0x2806058b4808ec83

```
sudo syscount -c -d 5
...
futex                56728
sched_yield        4439754
clock_gettime     32326537

erwincoumans, 2018-07-25:
Before disabling timings all together, we should remove the ones in the most innerloops. Let me look at it. Otherwise we can add some API to enable/disable timers.