bulletphysics/bullet3 #1420: Crash in example Unity script for C# bindings of PhysicsClientC_API

patrickDry, 2017-11-04:
Using both the latest release and the latest master, the provided example script showing usage of the C# bindings within Unity crashes.

The crash occurs with b3GetBodyInfo, and also with b3GetJointInfo.

The stack shows:
```
========== OUTPUTING STACK TRACE ==================

0x00007FFA9DD32CD3 (mono-2.0-bdwgc) [c:\buildslave\mono\build\mono\metadata\object.c:6264] mono_string_new_checked 
0x00007FFA9DD331E6 (mono-2.0-bdwgc) [c:\buildslave\mono\build\mono\metadata\object.c:6317] mono_string_new_wrapper 
  ERROR: SymGetSymFromAddr64, GetLastError: 'The specified module could not be found.' (Address: 00000000102760F8)
  ERROR: SymGetModuleInfo64, GetLastError: 'A dynamic link library (DLL) initialization routine failed.' (Address: 00000000102760F8)
0x00000000102760F8 ((<unknown>)) 
0x00000000365A7FD0 (Mono JIT Code) (wrapper managed-to-native) NativeMethods:b3GetBodyInfo (intptr,int,b3BodyInfo&)
0x00000000365A4C40 (Mono JIT Code) [C:\Users\drycactus\Documents\dev\pybullet-unity\Assets\Scripts\NewBehaviourScript.cs:93] NewBehaviourScript:Start () 
0x0000000036646D54 (Mono JIT Code) (wrapper runtime-invoke) object:runtime_invoke_void__this__ (object,intptr,intptr,intptr)
0x00007FFA9DB2434A (mono-2.0-bdwgc) [c:\buildslave\mono\build\mono\mini\mini-runtime.c:2618] mono_jit_runtime_invoke 
0x00007FFA9DD28BC7 (mono-2.0-bdwgc) [c:\buildslave\mono\build\mono\metadata\object.c:2853] do_runtime_invoke 
0x00007FFA9DD315CF (mono-2.0-bdwgc) [c:\buildslave\mono\build\mono\metadata\object.c:2905] mono_runtime_invoke 
0x000000014091D03C (Unity) scripting_method_invoke
0x00000001409119C3 (Unity) ScriptingInvocation::Invoke
0x00000001408DC82F (Unity) MonoBehaviour::InvokeMethodOrCoroutineChecked
0x00000001408DC996 (Unity) MonoBehaviour::InvokeMethodOrCoroutineChecked
0x00000001408DE928 (Unity) MonoBehaviour::Start
0x00000001408DF449 (Unity) MonoBehaviour::DelayedStartCall
0x00000001403A1FEA (Unity) DelayedCallManager::Update
0x00000001405DB34C (Unity) `InitPlayerLoopCallbacks'::`34'::EarlyUpdateScriptRunDelayedStartupFrameRegistrator::Forward
0x00000001405D9ADC (Unity) PlayerLoop
0x00000001410B9862 (Unity) PlayerLoopController::UpdateScene
0x00000001410C2FBA (Unity) PlayerLoopController::EnterPlayMode
0x00000001410C3829 (Unity) PlayerLoopController::SetIsPlaying
0x00000001410C4682 (Unity) Application::TickTimer
0x00000001412CC349 (Unity) MainMessageLoop
0x00000001412CDC04 (Unity) WinMain
0x0000000141DE45A8 (Unity) __tmainCRTStartup
0x00007FFADD448364 (KERNEL32) BaseThreadInitThunk
0x00007FFADF057091 (ntdll) RtlUserThreadStart

========== END OF STACKTRACE ===========
```

I have been unable to find a solution to this issue so far, any ideas?

patrickDry, 2017-11-04:
Was using old NativeMethod.cs autogenerated file, apologies

closed 2017-11-04 by patrickDry

erwincoumans, 2017-11-04:
Great, thanks! This is very experimental work, the bindings etc. If you do anything useful with it, or have some improvements/suggestions, please let me know (you can reply in this closed issue).

patrickDry, 2017-11-04:
Thanks for the input, I plan on porting the b3RobotSimulatorClientAPI to native C# to be used in a game context within Unity.
I'll let you know how it goes and if I find any major issues or required features.

patrickDry, 2017-11-08:
Hi Erwin,
a small issue so far, the NativeMethods.cs has incorrect parameter declaration in the function prototypes when a user needs to pass a double[].
```
// Example
public static extern int b3CreateCollisionShapeAddBox(IntPtr commandHandle, ref double halfExtents);
// Should be
public static extern int b3CreateCollisionShapeAddBox(IntPtr commandHandle, IntPtr halfExtents);
```
When calling, a user needs to allocate some memory and pass a pointer, like so:
```
double[] halfExtents = new double[] { 0.5, 0.5, 0.5 };
GCHandle handle = GCHandle.Alloc(halfExtents, GCHandleType.Pinned);
NativeMethods.b3CreateCollisionShapeAddBox(cmd, handle.AddrOfPinnedObject());
```

I also wanted to ask if it is possible to calculate inverse kinematics only for a portion of the robot or if it can only be calculated for the whole hierarchy?
We need to calculate IK for individuals arms of a robot, I haven't dived into it yet but just wanted to get some input first.

Lastly, I need to do more research into this, but when attempting to manually apply gravity to the base link of a robot I'm running into some issues, if I use a "robot" created with the b3CreateBoxShapeCommandInit approach it works fine, if I instead load a URDF (ex: cube.urdf), then as soon as the cube rotates the gravity is applied what seems to be locally and not in world coordinates anymore, and so far I've had little success in understanding why, this is what I use to apply gravity:
```
double[] force = new double[] { 0, -10, 0 };
double[] pos = new double[] { 0, 0, 0 };
GCHandle forceHandle = GCHandle.Alloc(force, GCHandleType.Pinned);
GCHandle posHandle = GCHandle.Alloc(pos, GCHandleType.Pinned);
IntPtr cmd = NativeMethods.b3ApplyExternalForceCommandInit(pybullet);
NativeMethods.b3ApplyExternalForce(cmd, bodyId, -1, forceHandle.AddrOfPinnedObject(), posHandle.AddrOfPinnedObject(), 1);
IntPtr status = NativeMethods.b3SubmitClientCommandAndWaitStatus(pybullet, cmd);
```

Any thoughts?
Thanks,
Patrick

AndresTraks, 2017-11-08:
Instead of "ref double", you can set the parameter type to "double[]" and pass the array directly.

erwincoumans, 2017-11-09:
> you can set the parameter type to "double[]" and pass the array directly.

Does that fix your issue? I'm not very familiar with C# myself.

>  to calculate inverse kinematics only for a portion of the robot 

The underlying IK can do it, but we need to expose the functionality: it can handle arbitrary tree-IK with multiple end effector targets, but we only exposed what we needed at the moment. We'll look into it.

b3ApplyExternalForceCommandInit has some option to apply forces in local or world frame.

You may want to create some user friendly C# wrapper,with a similar API to [pybullet](http://pybullet.org) is that what you work on?

patrickDry, 2017-11-09:
Thanks @AndresTraks for the tip, it does indeed work as expected by simply passing a double[] instead of having to use an allocated garbage collection handle, although I wonder if it has any memory implications.

@erwincoumans Great news that the IK solver can handle arbitrary tree-IK, we would love to have this in pybullet and the C# bindings as it's essential for the game prototype we're working on.
If you have any pointer on how to expose this ourselves I'd love to know.

b3ApplyExternalForceCommandInit works fine on simple shapes created with the b3CreateBoxShapeCommandInit command but behaves very strangely with URDF files, even the simple cube.urdf and applying a simple force along a world axis leads to some wild intermittent spinning, I have yet to fully understand the issue but it might be related to inertial values?

We are currently working on the game prototype while I experiment with pybullet/C#, in the long-term if we end up using it in production I'd like to do a friendly C# wrapper and put itin the examples folder of the bullet repo if appropriate.

One other question if I may, I can't fully understand how to create visual shapes without loading an external file? There are methods to create the collision shapes and multibodies, but nothing meaningful that I can find for visuals, is this yet to be done?

A small example of what we're doing, this is within Unity and using the simple position IK solver to move the arm: https://i.imgur.com/1LQsdbc.gif

Thanks,
Patrick

patrickDry, 2017-11-13:
The meaningful way of integrating this into Unity is to be able to create links/joints programmatically and not by loading URDF files.
As such I think a few vital functions are not exposed, namely when creating a multiBody there is no way of defining joint paramters, such as origins, limits, etc.
Is this another part that is yet to be implemented? Any tips on how we could potentially do the work on our end?

erwincoumans, 2017-11-13:
Yes, it is a todo. For our purposes we really want to load URDF/SDF files into Unity and create graphics objects, and not the other way around. The is a starting point for programmatic creation, and you can use 'changeConstraint' to adjust some parameters but not all. If/when you have made some progress, please share it and it can motivate us to add the needed features.

patrickDry, 2017-11-13:
Thanks for the reply.
Yes it seems we're approaching it from two different angles, we're doing a game prototype where players can design robots in-game and so obviously need to create bullet objects programmatically, as well as be able to solve arbitrary-tree IK which is essential to the game design :(
Creating a Unity-based URDF parser and simulator would be relatively trivial, particularly if it's limited to primitive shapes and doesn't have to load in the visual meshes, we've done half the work already so could get an example going in the next week or two.

In the meantime as we're unable to do certain vital tasks when creating multibodies programmatically, we'll try resorting to a "trick" we've been using until now, which is simply creating the URDF files programmatically, loading them up and then matching the link index with the correct mesh for transforms updating.

If we could get arbitrary-tree IK and full support for programmatic creation of bodies and constraints we could then, on our end, merge the two things together and create a Unity-based application that parses URDF files, allows players to modify the robots within the application, simulate, and export any changes back to URDF format, creating a sort-of robot sandbox builder "game".