bulletphysics/bullet3 #1429: Adding projection matrix changes view angle

virajmehta, 2017-11-10:
It seems that when we compute a view and projection matrix, the view is correct in renderings if we do not include the projection matrix and incorrect when we do. A minimal example is available at https://github.com/kuanfang/debug-bullet. If you call example.py with argument 0, it renders an image using just the view matrix, and if you call it with 1, it renders using view and projection matrices. Please let me know if there is a fix that allows us to use our intrinsics along the correct line of sight. Thanks!!

erwincoumans, 2017-11-11:
You cannot use numpy format matrices, it needs to be a flat list of 16 floats.  Convert it, for example:
 return projection_matrix.transpose().flatten()

Then, compare your computation with the one from Bullet, by printing it:
 print("projection_matrix",projection_matrix)

closed 2017-11-11 by erwincoumans

virajmehta, 2017-11-12:
I think that fixed our problem. Thanks so much!