bulletphysics/bullet3 #1434: Modelling a deformable softbody gripper in pyBullet

achuwilson, 2017-11-16:
Hi, I was wondering whether a deformable gripper like the GelSight gripper (linked below)can be simulated in pyBullet. 

[MIT News](http://news.mit.edu/2017/gelsight-robots-sense-touch-0605)
[Arxiv Paper](https://arxiv.org/abs/1708.00922)

I tried to model a tactile sensor using the getContactPoints API, but it returns only a couple of contact points when gripping objects, which may be attributed to the non-deformability of the solids. Is it possible to expose the softbody simulation capabilities of Bullet in python APIs. In addition, will the collision detection queries work with the softbodies.

ashwal, 2017-11-16:
You can take a look at issue #1432 and #1175 for some conversation around this, but no outward facing progress has been made.

I'm working on basic bindings right now.

JanMatas, 2017-12-29:
HI @ashwal 

have you made any progress on the bindings lately? I am looking for a way to simulate cloth grasping using pybullet. 

Thanks!

ashwal, 2017-12-30:
@JanMatas For my uses it ended up being faster just to write in C++ in hacky way. Sorry!

1seck, 2020-05-30:
Hi @ashwal 
could you just lay out in 5 steps how to do so?
New to python here :)
Thank you!

closed 2021-04-26 by erwincoumans