bulletphysics/bullet3 #1437: Connecting to GUI is extremely slow on some Ubuntu machines

kuanfang, 2017-11-17:
On some of the Ubuntu computers I am using, it is extremely slow to run simply:
`
import pybullet as p
p.connect(p.GUI)
`
It is stuck somewhere between Bullet prints "pthread_getconcurrency()=0" and "b3Printf: Selected demo: Physics Server". On the Ubuntu computer I am using, this constantly happens every time I run the code above. But it works fine on some other computers in our lab.

One difference between the problematic machine and the others which might cause the problem is that, the problematic machine has two GPUs (GeForce GTX 660 + GeForce GTX TITAN X). And when the ExampleBrowser starts, it always choose the inferior GTX 660 (as Bullet prints "GL_RENDERER=GeForce GTX 660/PCIe/SSE2"). I was wondering if there is a way to specify which GPU we want to use for Bullet.

erwincoumans, 2017-11-17:
It can be anything, so it will help giving some suggestions (some obvious ones):
+ Did you try rebooting your machine?
+ Remove that second GPU to see if that is really causing the issue
+ Update to better/latest Nvidia drivers? (there are proprietary ones etc)
+ Does the slow delay also happen if you start the example browser first, in server mode, and connect over shared memory?
+ build Bullet/pybullet using glfw with premake (cd build3, premake4_linux64 --enable_glfw --enable_pybullet  --double --python_include_dir="....." --python_lib_dir="....")

kuanfang, 2017-11-18:
Hi Erwin! I found the delay is caused by glClear [here](https://github.com/bulletphysics/bullet3/blob/master/examples/OpenGLWindow/X11OpenGLWindow.cpp#L983). And I also found that after restarting the computer, at the first time I run pybullet and connect to GUI, it goes smoothly. But starting with the second time, the delay is significant (more than a minute). 

Meanwhile, among your suggestions, `p.connect(p.GUI,options="--opengl2")` and `./App_BulletExampleBrowser_gmake_x64_release --start_demo_name="Physics Server"` does not solve the problem. The delay is the same in both situations.

closed 2017-12-14 by kuanfang