bulletphysics/bullet3 #1438: Softbody each face is separated

AndreaCatania, 2017-11-17:
Hi @erwincoumans I've a problem with softbody implementation and I don't know how fix this, check this video: https://youtu.be/Dr79SCBkgxQ

As you can see at the right side I've a sphere that should be a Softbody but each face is separeted and all faces crash to the ground... It doesn't create a volume.

Instead in the left side you can see a very good simulation of a cloak.

I'm using the same code to initialize these, but when I submit a "volumed" mesh this is the behaviour.

The code that createthe softbody is this https://github.com/AndreaCatania/godot/blob/00f7518c934e7229df7eb82d41cff99ed3ff3e0c/modules/bullet/soft_body_bullet.cpp#L410 

and this is the initialization: https://github.com/AndreaCatania/godot/blob/00f7518c934e7229df7eb82d41cff99ed3ff3e0c/modules/bullet/soft_body_bullet.cpp#L433-L472

What could be the cause of this problem?

erwincoumans, 2017-11-17:
Vertices with the same position need to be shared, in order to be connected for softbodies. You can create some mappings, if your have multiple vertices with the same position.

erwincoumans, 2017-11-17:
Aside from that, you may need to generate extra edges, etc. Study the softbody demos (the code) in the Bullet Example Browser.

erwincoumans, 2017-11-18:
There is some generateBendingConstraints helper, and btSoftBodyHelpers.

>> pressure = 0 and mass = 1

For soft body settings and features, please study the softbody demos (and the [C++ code](https://github.com/bulletphysics/bullet3/blob/master/examples/SoftDemo/SoftDemo.cpp)) in the Bullet Example Browser.

For another source of inspiratoin, you could check out Blender, it has some [conversion for btSoftBody ](https://github.com/dfelinto/blender/blob/3068ea34e414e7af6d20af7ab92c69cbd9a33414/source/gameengine/Physics/Bullet/CcdPhysicsController.cpp#L264)too for some features.

Notice things like btSoftBody:;updateBounds (needs to be called at the right time),  shape matching features (setPose) , generateClusters etc. You may want to expose those as options in Godot Engine, since those features are not universally applicable.

If you are still stuck after that, please ask another specific question. Good luck!

AndreaCatania, 2017-11-18:
Thanks you for your response, I've fixed this issue. Now the implementation is quite completed but I've this last problem.

https://www.youtube.com/watch?v=xJciXWShCr8&feature=youtu.be

As you can see in the begining of the video in the editor the sphere normals are correctly outside the mesh: 
[image]

When I start the simulation the mesh normals are inverted:
[image]

I'm searching what cause it, Can you imagine what's the cause is?

However Thanks you for help!

AndreaCatania, 2017-11-18:
I fixed it, by inverting the indices order. This is odd why bullet require CCW indices?

erwincoumans, 2017-11-18:
>> bullet [softbody] require CCW indices?

Who knows, the original creator of btSoftBody has long gone, that code is over 10 years old.

It will take some time to understand the limitations and possibilities of btSoftbody, and exposing the various options will help (shape matching enabled/disabled, collision clusters ON/OFF, generating bending constraints etc.).

AndreaCatania, 2017-11-18:
Oh ok, I'll expose it. In the main while if you want see how look like in this moment: https://www.youtube.com/watch?v=8A6PYFSesKQ

closed 2017-11-22 by AndreaCatania