bulletphysics/bullet3 #1441: sudo pip3 install pybullet error ubuntu 16.04

Robomate, 2017-11-17:
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/home/roboball/.local/lib/python3.5/site-packages/numpy/core/include -Iexamples/ThirdPartyLibs/optionalX11 -Isrc -Iexamples/ThirdPartyLibs -Iexamples/ThirdPartyLibs/Glew -Iexamples/ThirdPartyLibs/enet/include -Iexamples/ThirdPartyLibs/clsocket/src -I/usr/include/python3.5m -c src/BulletDynamics/Vehicle/btRaycastVehicle.cpp -o build/temp.linux-x86_64-3.5/src/BulletDynamics/Vehicle/btRaycastVehicle.o -DGWEN_COMPILE_STATIC -DBT_USE_DOUBLE_PRECISION -DBT_ENABLE_ENET -DBT_ENABLE_CLSOCKET -DB3_DUMP_PYTHON_VERSION -DPYBULLET_USE_NUMPY -D_LINUX -DGLEW_STATIC -DGLEW_INIT_OPENGL11_FUNCTIONS=1 -DGLEW_DYNAMIC_LOAD_ALL_GLX_FUNCTIONS=1 -DDYNAMIC_LOAD_X11_FUNCTIONS -DHAS_SOCKLEN_T -fno-inline-functions-called-once
    cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
    src/BulletDynamics/Vehicle/btRaycastVehicle.cpp: In member function ‘virtual void btRaycastVehicle::updateFriction(btScalar)’:
    src/BulletDynamics/Vehicle/btRaycastVehicle.cpp:716:1: internal compiler error: Segmentation fault
     }
     ^
    Please submit a full bug report,
    with preprocessed source if appropriate.
    See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-5hmlxw8u/pybullet/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-wawqnotv-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-5hmlxw8u/pybullet/

Robomate, 2017-11-17:
not sure why this happened try it again and it works

closed 2017-11-17 by Robomate

erwincoumans, 2017-11-18:
Odd, seems some compiler bug. By the way, we could safely remove btRaycastVehicle.cpp from pybullet, since we don't use/expose it in pybullet.

JobSmith, 2018-11-11:
However, this error still occurs. Have you removed it yet?