bulletphysics/bullet3 #1444: Specifying the maximum joint velocities in the position control.

kuanfang, 2017-11-18:
Currently there is no arguments in the position control for specifying the maximum joint velocity when controlling the robot arm, which is an important feature in many real-world robot APIs. I thought the pybullet might take the velocity limit specified in the URDF file of the robot. However, I found changing the velocity limit in the URDF does not make any differences, in terms of the time it takes to control the arm to  aset of specified target positions. I even tried to set the velocity limit to be all 0, and the arm still moves really fast.

I found setting maximum joint velocity would be an important feature. For example, the Intera SDK of the Sawyer robot lets user specify the maximum joint velocity in position control, and set it to be 0.3 of the maximally allowed joint velocity. Currently, without setting the maximum joint velocity, the Sawyer robot moves to any position in around 0.01~0.05 sec, while in the real world it takes 1~5 sec, which is very unrealistic. 

I was wondering what is the current way for Bullet to set the joint velocity during position control? Is there any way for us to specify the maximum velocity? If not, would it be possible for Erwin and Yunfei to add such a feature for more realistic robot control?  Thanks.

erwincoumans, 2017-11-20:
Yes, we can look into adding support for maximum velocity control (due to the motor).

erwincoumans, 2017-11-22:
Feature is implemented here: https://github.com/bulletphysics/bullet3/pull/1449

closed 2017-11-22 by erwincoumans