bulletphysics/bullet3 #1446: Velocity integration is wrong

d3x0r, 2017-11-19:
I was analyzing the behavior or cannon.js ( https://github.com/schteppe/cannon.js/issues/353 ) and learned that the velocity step across time should be  `P += ( Vi + Vf ) / 2`  and not `P += Vf`.

Unfortunately this isn't a simple fix for bullet, because you don't keep the initial velocity (the one at the begin of a time frame) to later apply...  velocity is updated in `body->integrateVelocities( timeStep);`

and then after much work is applied applied

```
				body->predictIntegratedTransform(timeStep, predictedTrans);
				body->proceedToTransform( predictedTrans);
```

which eventually calls 

```
		predictedTransform.setOrigin(curTrans.getOrigin() + linvel * timeStep);
```

but `linvel` should be `(initialLinVel + linvel)/2`

At least according to 
http://www.physicsclassroom.com/class/1DKin/Lesson-6/Kinematic-Equations

a similar modification should be made for the quaternion math in that same routine (which it sort of looks like someone has tried to do, but I'm sure it can be simplified) that or the `btScalar(0.5)` makes all rotations half speed.

and testing with a similar modification that made a simple falling object much more accurate in cannon.

Instantaneous forces... unfortunately all get grouped together as constant acceleration over time.

The next thing I found in cannon though, was that the damping force application causes progressive errors at different time steps...  don't have a solution for that though.

erwincoumans, 2017-11-19:
Bullet uses the semi-implicit Euler integration, as described here: https://en.wikipedia.org/wiki/Semi-implicit_Euler_method. One of the benefits is that semi-implicit integration almost conserves the energy.

As the time step gets smaller, it will approximate the analytical solution, but at larger time steps there is an approximation error indeed.

closed 2017-11-19 by erwincoumans

d3x0r, 2017-11-19:
The other method (including a div by 2) is more conservative of energy, and more accurate between varying tick rates.
(clients running at 90fps and servers running at 30 or less for instance).

For a falling body, with your existing method, the body will fall further than it should, thereby gaining energy; and causing guaranteed desync between different systems.