bulletphysics/bullet3 #1450: Building with NMake (VS2017) produces warning D9025 : overriding '/arch:SSE' with '/arch:SSE2'

avikde, 2017-11-22:
My system:
* Windows 10 with latest updates
* Installed the VS2017 build tools from https://www.visualstudio.com/downloads/#build-tools-for-visual-studio-2017
* cmake 3.9.6 from https://cmake.org/download/ (latest)

Steps:
* pulled the latest source, opened a developer command prompt for VS2017, and roughly followed the bat file:

```
> cd bullet3
> mkdir build_cmake
> cd build_cmake
> cmake .. -DBUILD_PYBULLET=OFF -DUSE_DOUBLE_PRECISION=ON -DCMAKE_BUILD_TYPE=Release -G "NMake Makefiles"
> nmake
```

Output (for each compilation line)
```
[ 20%] Building CXX object examples/ThirdPartyLibs/BussIK/CMakeFiles/BussIK.dir/Misc.obj
cl : Command line warning D9025 : overriding '/arch:SSE' with '/arch:SSE2'
Misc.cpp
```

The libraries build, and the examples run fine, only the warning is annoying.

I tried googling but don't understand enough about VS with cmake to know how to fix it. I also tried `cmake .. -DUSE_MSVC_SSE2=ON -DUSE_MSVC_AVX=OFF ....` but that had the same result.

erwincoumans, 2017-11-22:
You could try to use USE_MSVC_SSE=OFF?
It looks like a silly warning, https://stackoverflow.com/questions/1067630/sse2-option-in-visual-c-x64

avikde, 2017-11-22:
That fixed it! Not sure why I didn't try that before.  For reference,
```
> cmake .. -DUSE_MSVC_SSE=OFF <other options> -G "NMake Makefiles"
```

closed 2017-11-22 by avikde