bulletphysics/bullet3 #1454: Unable to install pybullet with numpy enabled

jannikb, 2017-11-24:
I'm still getting the error described in #1411.

Using `pip install` or `python setup.py install`  the installation fails with the attached output. If i manually remove `-DPPYBULLET_USE_NUMPY` from the setup.py the installation works without errors. Before I was running pybullet 1.2.1 and had no problems installing it.

I'm running Windows 10, python 2.7.14 and numpy 1.13.3. 

[install.txt](https://github.com/bulletphysics/bullet3/files/1502358/install.txt)

erwincoumans, 2017-11-24:
Fixed here, can you try pip install -U pybullet, using 1.7.2?
https://github.com/bulletphysics/bullet3/pull/1455

closed 2017-11-24 by erwincoumans

erwincoumans, 2017-11-25:
actually, another fix was needed, please try pybullet 1.7.3 (using pip install -U pybullet now)

jannikb, 2017-11-25:
Thanks, 1.7.3 installed without any issues.