bulletphysics/bullet3 #1457: Getting the true depth value from the OpenGL depth buffer

kuanfang, 2017-11-25:
@andreykurenkov and I found that the depth rendering in pybullet is problematic at the moment. But to fix this issue, just two more lines of code are needed. This [link](https://stackoverflow.com/questions/6652253/getting-the-true-z-value-from-the-depth-buffer) explains the solution pretty well.

erwincoumans, 2017-11-25:
please submit a pull request, thanks a lot!

erwincoumans, 2017-11-25:
Please provide more information, since "problematic" is not clear enough.
getCameraImage can render using TinyRenderer on CPU and using OpenGL 3.x hardware, so there are two different code paths. Both should give similar results.

Please create a small test that recreates the problem, mention if the issue is related to CPU and/or OpenGL GPU rendering, and if there are just two lines needed to fix it, please open a pull request with those two lines.
Thanks a lot!