bulletphysics/bullet3 #1460: Setting time step to 0.001 or 0.01 makes the robot bounce off the ground and float.

neveryoubetray, 2017-11-26:
I am trying to use pybullet for a reinforcement learning algorithm (DDPG) using a custom urdf model I have made of a biped (12 joints and torso). I initialize my environment with gravity, then p.setTimeStep(DT), and step through the simulation at each time step by sending actions with p.setJointMotorControlArray(self.Id,order,p.POSITION_CONTROL, actions), then p.stepSimulation(). I am wondering if there is a limit for DT to give it time to run the simulation? I have tried DT of 0.001 and 0.01 and the robot seems to bounce off the ground plane and then float through the air like it is in water. When I run DT of 0.1, the feet ground (though seem to sink into the ground plane) and the simulation behaves as I might expect with the robot falling quickly over. Can I run the simulation quicker? Am I missing any steps in my implementation? Thanks for your time!

erwincoumans, 2017-11-30:
Never seen anything like this, it could be self-collision.

If you can provide a simple reproduction case (a pybullet file and URDF etc), someone may look into it.

neveryoubetray, 2017-12-01:
ah yep, sorry about that. I appreciate you taking the the time to respond. 

This repository contains my urdf under 'assets', I created it from a xacro using a ROS command. my gym-like environment is called 'env.py'. The rest of the code is from OpenAI baselines: 
https://github.com/neveryoubetray/ddpg

Thanks again.

ojasjoshi, 2020-09-24:
@erwincoumans  I see that you are mentioning delta time in Hz. Does that mean that the real-time-factor(RTF) for pybullet simulations is always(or by default) = 1. Or is the update rate(and hence the RTF) configurable via maxSolverIterations parameter?

erwincoumans, 2020-09-24:
This topic is unrelated to 'real-time factor'. Closing out-of-date issue.

closed 2020-09-24 by erwincoumans