bulletphysics/bullet3 #1469: b3LoadUrdf joint_properties tag ignored

avikde, 2017-12-06:
In my urdf, if I put
```
  <joint name="10" type="revolute">
    <parent link="body"/>
    <child link="hip2"/>
    <axis xyz="1 0 0"/>
    <origin xyz="0.3 -0.1 0"/>
    <limit effort="40" velocity="45" lower="-1.57" upper="1.57"/>
    <joint_properties damping="10.0" friction="10.0"/>
  </joint>
```

and then call
```C
		b3GetJointInfo(kPhysClient, b3id, jointId, &jointInfo);
		printf("%d\tdamping=%.2f, friction=%.2f\n", i, jointInfo.m_jointDamping, jointInfo.m_jointFriction);
```
the results are always 0.

Since in #1459 you pointed out the `URDF_USE_INERTIA_FROM_FILE` flag... I was wondering if there is a similar one for damping/friction that I'm missing?

wxmerkt, 2017-12-06:
@avikde I haven't seen ``joint_properties`` as an official tag - ``<dynamics>``, on the other hand is listed/official: http://wiki.ros.org/urdf/XML/joint - not sure whether that one is picked up though.

avikde, 2017-12-06:
@wxmerkt after a quick test it seems like it is both picked up from the urdf, as well as used in the physics solver! Thanks for your help.

I.e. the solution is to use
```
  <joint name="10" type="revolute">
    <parent link="body"/>
    <child link="hip2"/>
    <axis xyz="1 0 0"/>
    <origin xyz="0.3 -0.1 0"/>
    <limit effort="40" velocity="45" lower="-1.57" upper="1.57"/>
    <dynamics damping="1.0" friction="0.0"/>
  </joint>
```

closed 2017-12-06 by avikde