bulletphysics/bullet3 #1470: Soft body Pybullet bindings result in segmentation faults

dvercaut, 2017-12-07:
I have created basic pybullet bindings for loading Soft Bodies from an object file. However when running the example in the pybullet GUI (using DIRECT there is no problem), the simulation crashes after a random amount of time due to a segmentation fault (see trace below). I have created these bindings using the latest version of bullet and applied the Soft body patch that was referenced in #1432. And I am running the simulation on an Ubuntu 16.04.3 machine. Any help to solve this problem would be greatly appreciated.
(The code: https://github.com/adverley/bullet3/tree/soft-bodies-pybullet)

========== OUTPUTING STACK TRACE ==================
 0x00007ffff6510646 in btAlignedObjectArray<btAlignedObjectArray<btVector3FloatData> >::expand(btAlignedObjectArray<btVector3FloatData> const&) () from /home/dvercaut/ugent_git/bullet3/build_cmake/examples/pybullet/pybullet.so
0x00007ffff65129d6 in MultithreadedDebugDrawer::drawLine(btVector3 const&, btVector3 const&, btVector3 const&) () from /home/dvercaut/ugent_git/bullet3/build_cmake/examples/pybullet/pybullet.so
0x00007ffff5cb852e in btSoftBodyHelpers::Draw(btSoftBody*, btIDebugDraw*, int) () from /home/dvercaut/ugent_git/bullet3/build_cmake/src/BulletSoftBody/libBulletSoftBody.so.2.87
0x00007ffff651d61e in PhysicsServerCommandProcessor::renderScene(int) () from /home/dvercaut/ugent_git/bullet3/build_cmake/examples/pybullet/pybullet.so
0x00007ffff650998d in PhysicsServerExample::renderScene() () from /home/dvercaut/ugent_git/bullet3/build_cmake/examples/pybullet/pybullet.so
0x00007ffff610d102 in OpenGLExampleBrowser::update(float) () from /home/dvercaut/ugent_git/bullet3/build_cmake/examples/ExampleBrowser/libBulletExampleBrowserLib.so.2.87
0x00007ffff64e154a in ExampleBrowserThreadFunc(void*, void*) () from /home/dvercaut/ugent_git/bullet3/build_cmake/examples/pybullet/pybullet.so
0x00007ffff65d5740 in threadFunction(void*) () from /home/dvercaut/ugent_git/bullet3/build_cmake/examples/pybullet/pybullet.so
0x00007ffff77e46ba in start_thread (arg=0x7ffff3c3f700) at pthread_create.c:333
0x00007ffff6e0a3dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
========== END OF STACKTRACE ===========

erwincoumans, 2017-12-14:
It isn't supported yet, so please track down the crash. Also,in case someone wants to help out: provide the full patch you used and the exact script you ran, and how long it takes until the crash happens. Thanks, Erwin

dvercaut, 2017-12-17:
As mentioned the patch that was used was the one referenced in #1432 
[softbody_patch_clean.patch.zip](https://github.com/bulletphysics/bullet3/files/1565257/softbody_patch_clean.patch.zip)

The script in question is bullet3/examples/pybullet/examples/test_loadSoftBody.py, the crash seems to only happen when there is an interaction between a rigid body and a soft body. 
There is no exact moment when the crash happens, but in most cases the segmentation fault will occur after about 10 seconds.

JanMatas, 2018-01-05:
Hi @dvercaut , I have tried checking out your branch and I seem to be also getting random segmentation faults. I was wondering if you had any updates on this lately.

dvercaut, 2018-01-06:
I haven't made a lot of progress, but I plan to continue working on it a few weeks.
If you would like to help, I was planning on doing the following:

If you run the ExampleBrowser you'll see that I have added some examples for Softbodies. These examples use different underlying C code than the pybullet examples but in general they simulate the same thing. These examples however do not use the PhysicsServerCommandProcessor code, which is used by pybullet.

So the idea would be to create a new example in the ExampleBrowser that uses the same code as the pybullet examples (uses the code from PhysicsServerCommandProcessor), and see whether it also crashes. If this works we can at least rule out some possible causes.

erwincoumans, 2019-12-18:
Closing out-of-date issues. We are still developing soft body and deformables.

closed 2019-12-18 by erwincoumans