bulletphysics/bullet3 #1472: Your site is down

AndreaCatania, 2017-12-09:
Bullet site is not working:
http://bulletphysics.org/mediawiki-1.5.8/index.php/Constraints
http://bulletphysics.org

erwincoumans, 2017-12-09:
Thanks for the report. I'm in the process of moving it all to
https://pybullet.org/wordpress. Can you try it?
(not the wiki, just phpBB3 and wordpress)

erwincoumans, 2017-12-09:
The wiki page is up online now. I'll try to migrate the wiki page to the
new server too, and shutdown the old machine.
Can you try the pybullet.org server?

AndreaCatania, 2017-12-09:
It's working perfectly now, but lose your google ranks is not a problem?

closed 2017-12-12 by AndreaCatania

Gollum999, 2019-01-22:
Has the wiki moved to a new location?  Old links like https://pybullet.org/mediawiki-1.5.8/index.php/Collision_Callbacks_and_Triggers have been 404ing ever since the primary site moved.

DanBreton, 2019-03-06:
I'm also not able to find the wiki anymore. Would be great if it could be rehosted or given as a download? There was some really useful information on there.

sapiensl, 2019-05-27:
Hi! Any news on this issue? I am specifically interested in https://pybullet.org/mediawiki-1.5.8/index.php?title=Scaling_The_World and can not find a substitute anywhere as well as in the Wayback Machine... I would be so glad if the information somehow resurfaced!

dav64, 2019-11-10:
I suggest to reopen this issue as the wiki is still not up
Any plans to relocate it online?

@sapiensl thanks for wayback machine idea, thanks

ompadu, 2020-12-06:
one year later it's still down. web.archive helped, but it's not ideal.

jdumas, 2021-05-19:
+1 to this. The content of https://pybullet.org/mediawiki-1.5.8/index.php?title=Scaling_The_World is not available anywhere anymore, not even on webarchive...

sapiensl, 2021-05-25:
Hi, I have been summoned to return with a new working link to the "Scaling the World" article: https://web.archive.org/web/20170713085948/http://www.bulletphysics.org/mediawiki-1.5.8/index.php/Scaling_The_World
That is all I can do for you. Have a nice day!

Sky10086, 2022-09-29:
> Hi! Any news on this issue? I am specifically interested in https://pybullet.org/mediawiki-1.5.8/index.php?title=Scaling_The_World and can not find a substitute anywhere as well as in the Wayback Machine... I would be so glad if the information somehow resurfaced!
> 
> EDIT: Just as I wrote this, i found a (suboptimal) way of accessing it: https://web.archive.org/web/20170706235814/http://www.bulletphysics.org/mediawiki-1.5.8/index.php/Main_Page
> 
> Hope this helps anyone who might stumble upon this on the search.

Thank you so much! this help me a lot