bulletphysics/bullet3 #1473: Contact issues (sliding, instability) after cone friction update

avikde, 2017-12-11:
In my programs, a simple quadruped began sliding on the ground after I updated to the most recent master branch, whereas 18e7c30 works fine. 

You can replicate this by just running App_RobotSimulator.exe, with the latest (where it actually flips over, presumably because of a contact instability) and 18e7c30 (where it falls to the ground and stands).

My system is Windows 10; please let me know if I can provide any other information.

erwincoumans, 2017-12-12:
There is a way to disable cone friction, can you see if that helps?
C-API: b3PhysicsParamSetEnableConeFriction(command, 0  /*enableConeFriction*/);
pybullet: p.setPhysicsEngineParameter(enableConeFriction=0)

avikde, 2017-12-12:
Yes, that resolved it! Thanks for your prompt help. Presumably this will also fix RobotSimulator.

erwincoumans, 2017-12-12:
thanks for testing it. If I find time I'll track down why cone friction
adds sliding in this case, it shouldn't.

avikde, 2017-12-12:
I tried for a bit to look into the issues, but b031ba0 looks like a huge update that I don't see how to chunk into smaller pieces. It seems like the updates to btMultiBodyConstraintSolver in that commit are the relevant ones? If you have time to write a paragraph or two on what the cone friction update means and is supposed to do I can try to dig deeper, time permitting.

Also I guess this should be resolved (from your comment) so I reopened the issue to keep track of the fixes / development

erwincoumans, 2017-12-14:
It should be fixed in this commit, the bug was that we accumulated the residual, instead of squared residual. https://github.com/bulletphysics/bullet3/pull/1479/commits/5ab6a0ad434447d99b3080ec9a86a1f733a5b2cb

The cone friction is described in slide 31 of my SimPar 2016 presentation:
https://docs.google.com/presentation/d/1q00u8IY1biDcVwkhfXaJrurspk35NrwPTBINeRKm2c8/edit#slide=id.p9
Instead of approximating the friction cone with a polygon (pyramid) we clamp both friction directions simultaneously against the actual cone.

erwincoumans, 2017-12-14:
Note that the quadruped.urdf used in App_RobotSimulator is very old before we did system identification. We made a lot of progress simulating the Minitaur and its motors etc, so that the transfer from simulation to real hardware is starting to work now. I'll make some updated demo soon, with controls to walk forward/turn. 

You can also enable friction_anchors in the contact section. Attached is an example (still a version without proper system identification). friction_anchors will remove all drift in the tangential friction.

[minitaur.zip](https://github.com/bulletphysics/bullet3/files/1560458/minitaur.zip)

avikde, 2017-12-14:
Thanks for the fix, and also the tip about friction_anchors - I'll try it out

closed 2017-12-14 by avikde

erwincoumans, 2017-12-14:
You can add this section to the legs / links that touches the ground:

`<contact>
			<friction_anchor/>
			<stiffness value="3000.0"/>
			<damping value="100.0"/>
			<spinning_friction value=".3"/>
			<lateral_friction value="1"/>
		</contact>`