bulletphysics/bullet3 #1477: Joint angular limits always word space

AndreaCatania, 2017-12-12:
Hi, I searched across forum and web but I didn't found an answer to this problem, I don't know why but the joints are in word space, check the gif below

[image]

As you can see the angular limits doesn't take care of frame transformation.

This is how bullet works or I have  to search the bug elsewhere?

erwincoumans, 2017-12-12:
There are always two bodies (and frames) involved with a btGeneric6DofSpring2Constraint (rbA and rbB, frameA and frameB), and the joint limits are expressed in parent frame (rbA, frameA).

Did you already move to use btGeneric6DofSpring2Constraint?

For btMultiBody (which is recommended for perfectly stiff joints, such as doors or characters/robots driven by strong motors and IK), the joint limits are also expressed in parent/joint frame.

AndreaCatania, 2017-12-12:
oh, Thanks you!

>> Did you already move to use btGeneric6DofSpring2Constraint?

I didn't it yet, but it's something to do to fully implement bullet

closed 2017-12-12 by AndreaCatania