bulletphysics/bullet3 #1480: Is it possible to use the discrete GPU for rendering instead of the integrated one?

avikde, 2017-12-19:
This is more of a query than a request: I have a laptop with integrated as well as discrete GPU (Windows 10) and bullet3 only uses the integrated one. Is there a way to make it use the discrete GPU when available? MATLAB (for e.g.) uses the dGPU when it opens so I think it's possible for the program to request or select it.

If you could point me to where in the OpenGL init code I should dig, I can look myself as well.

Thanks!

erwincoumans, 2017-12-19:
It is not enabled indeed, our home-grown implementation is pretty simple.
The OpenGLWindow implementation is all here:
https://github.com/bulletphysics/bullet3/blob/master/examples/OpenGLWindow/Win32OpenGLWindow.cpp

Recently I added some option to enable GLFW, it could be worth trying out.

It is only enabled with premake4, see the build3/premake4.lua file.

newoption
    {
trigger     = "glfw_include_dir",
value       = default_glfw_include_dir,
description = "GLFW 3.x include directory"
    }

newoption
    {
                        trigger     = "glfw_lib_name",
                        value       = default_glfw_lib_name,
                        description = "GLFW 3.x library name (glfw, glfw3)"
    }

    newoption
    {
trigger     = "glfw_lib_dir",
value       = default_glfw_lib_dir,
description = "(optional) GLFW 3.x library directory "
    }

    newoption
    {
trigger     = "enable_glfw",
value       = false,
description = "(optional) use GLFW 3.x library"
    }

if _OPTIONS["enable_glfw"] then
defines {"B3_USE_GLFW"}

So if GLFW has that feature, you may use that, or backport it to Bullet
OpenGL.

Let me know how it goes, or if you need some further help,
Thanks!
Erwin

avikde, 2017-12-20:
I found a solution for NVidia specifically https://docs.nvidia.com/gameworks/content/technologies/desktop/optimus.htm and created a pull request that makes the request on windows.

avikde, 2017-12-22:
Solution found within user code, so I'm closing this

closed 2017-12-22 by avikde