bulletphysics/bullet3 #1485: Replicating analytic CartPole with Bullet

rhaps0dy, 2017-12-23:
I'm trying to make a Bullet CartPole that accurately matches another CartPole that I have with explicit differential equations, for use with the [Roboschool](http://github.com/openai/roboschool). One of the things that I found was that the explicit moments of inertia set in my [MJCF File](https://gist.github.com/rhaps0dy/6e0906e747ffb913c678e1bbab3a0823) were completely ignored, which I fixed in #1484 . Now, there is another source of difference that I cannot find.

This is a plot of the 4-dimensional state space (Y axes) over time (X axes), without applying any torque and with the initial state specified in the comment in the MJCF file. The two plots at the left refer to the horizontal motion of the cartpole (`slider` in the linked MJCF), the top one represents position and the bottom one velocity. The two plots at the right refer to the pole's rotation (`hinge` in the MJCF), again top is angle and bottom is angular velocity.

The green line is analytic without friction in `slider`, the orange line adds some friction to it. Then Bullet stands in the middle of them. However, Bullet is supposed to have no friction! `frictionloss` in the MJCF file isn't parsed and I've added no motors with target 0.

Where does this loss of velocity come from? Any ideas? I'm using roboschool, which AFAIK just calls `loadMJCF` from the PhysicsC_Client. Admittedly this latter library isn't compiled and installed by CMake in `master`, only in [my fork](https://github.com/rhaps0dy/bullet3/commit/fc65beb970830b607ae11c8ec6912eae1bfb52c5).

[image]

[Image link in case the above doesn't work](https://imgur.com/a/PlVzH)

It should be possible to test the MJCF file by dropping it in as a replacement to [roboschool/roboschool/mujoco_assets/inverted_pendulum.xml](https://github.com/openai/roboschool/blob/master/roboschool/mujoco_assets/inverted_pendulum.xml) and setting the initial state in [gym_pendulums.py](https://github.com/openai/roboschool/blob/master/roboschool/gym_pendulums.py), I can make a mostly self-contained generator for the plot above if interested.

erwincoumans, 2017-12-23:
There is some default linear and angular damping in btMultiBody. You can disable it using pybullet

pybullet.changeDynamics(bodyUniqueId=..., linkIndex=..., linearDamping=0, angularDamping=0)
Do this for the base (linkIndex=-1) and all links.
See also the [PyBullet quickstart guide](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#).

Why do you want to use Roboschool instead of PyBullet?
PyBullet is more actively developed, much easier to install, and it exposes the full PhysicsClient_CAPI.

rhaps0dy, 2017-12-23:
>Why do you want to use Roboschool instead of PyBullet?

Ignorance. I didn't know PyBullet had essentially the same things but with one less wrapper to worry about and no outdated Bullet fork. Thank you!

The modification to the MJCF importer also affects PyBullet though.

I read the default damping parameters that you linked. That looks exactly like what I need. Should I, and how would I, add support for loading those from MJCF/URDF? Is the procedure related to this https://github.com/bulletphysics/bullet3/blob/master/src/BulletDynamics/Featherstone/btMultiBodyLink.h#L147-L148 ?

erwincoumans, 2017-12-23:
> I didn't know PyBullet

It would be nice if PyBullet got some proper publicity indeed.

> The modification to the MJCF importer also affects PyBullet though.

I prefer to maintain backward compatibility, and not starting to read new fields that were ignored before. So we would need to add flags like URDF_USE_INERTIA_FROM_FILE for MJCF, and URDF_USE_DAMPING_FROM_FILE etc. It seems easier to just add a line in pybullet for the linear/angular damping.

Note that the current linear/angular damping for btMultiBody is not joint damping: the velocity of each body and link is dampened. So in the case of a body that consists only of a base and no child links/joints, there is still damping.

The body linear/angular velocity damping is implemented here:
~~~~
for the base:
https://github.com/bulletphysics/bullet3/blob/master/src/BulletDynamics/Featherstone/btMultiBody.cpp#L826
and for child links:
https://github.com/bulletphysics/bullet3/blob/master/src/BulletDynamics/Featherstone/btMultiBody.cpp#L913
~~~~
This body damping can only be changed using pybullet.changeDynamics.

Separately, there is some explicit joint damping applied here:
https://github.com/bulletphysics/bullet3/blob/master/examples/SharedMemory/PhysicsServerCommandProcessor.cpp#L5868

This joint damping is read from the URDF file here:
https://github.com/bulletphysics/bullet3/blob/8c04bc9473b4c179d9781fc3577910354a2d9d84/examples/Importers/ImportURDFDemo/UrdfParser.cpp#L1095

rhaps0dy, 2018-01-18:
Hello,

Using the information in your latest comment, I was able to fix all my problems and replicate the particular cartpole I had in Bullet. Thank you very much!

To do that, though, I have added parsing of the `damping` MJCF attribute into `m_jointDamping`, as in here https://github.com/rhaps0dy/bullet3/commit/6995995a0bfa5963d313c34043cf3bac41393e48. Would you be interested in a pull request with this, adding a `URDF_USE_DAMPING_FROM_FILE` or something?

Thank you for your excellent work and help, and sorry it's been such a long time.

erwincoumans, 2019-12-18:
Thanks for the report. I would just use the pybullet.changeDynamics instead to set link damping and joint damping. Closing for now.

closed 2019-12-18 by erwincoumans