bulletphysics/bullet3 #1486: C API JointState m_jointMotorTorque is 0

avikde, 2017-12-23:
As a simple example, in https://github.com/bulletphysics/bullet3/blob/master/examples/TwoJoint/TwoJointMain.cpp I printed state.m_jointMotorTorque, like 

```
		// apply some torque
		b3GetJointInfo(kPhysClient, twojoint, jointNameToId["joint_2"], &jointInfo);
		command = b3JointControlCommandInit2(kPhysClient, twojoint, CONTROL_MODE_TORQUE);
		b3JointControlSetDesiredForceTorque(command, jointInfo.m_uIndex, 0.5 * sin(10*simTimeS));
		statusHandle = b3SubmitClientCommandAndWaitStatus(kPhysClient, command);

		// get joint values
		command = b3RequestActualStateCommandInit(kPhysClient, twojoint);
		statusHandle = b3SubmitClientCommandAndWaitStatus(kPhysClient, command);
		b3GetJointState(kPhysClient, statusHandle, jointNameToId["joint_1"], &state);
		q[0] = state.m_jointPosition;
		v[0] = state.m_jointVelocity;
		u[0] = state.m_jointMotorTorque;
		b3GetJointState(kPhysClient, statusHandle, jointNameToId["joint_2"], &state);
		q[1] = state.m_jointPosition;
		v[1] = state.m_jointVelocity;
		u[1] = state.m_jointMotorTorque;

		statusHandle = b3SubmitClientCommandAndWaitStatus(kPhysClient, b3InitStepSimulationCommand(kPhysClient));

		// debugging output
		printf("%.3f\t%.3f\t%.3f\t%.3f\t%.3f\t%.3f\t%.3f\n", simTimeS, q[0], q[1], v[0], v[1], u[0], u[1]);
		b3Clock::usleep(1000.*1000.*FIXED_TIMESTEP);
```

However the jointMotorTorque is always 0 even though clearly some joint torque is applied. Is this a mistake I'm making in using the C API? I read the pybullet docs about this and looked at `b3GetJointState()` in `PhysicsClientC_API.cpp` and it looks like it shouldn't be 0.

erwincoumans, 2017-12-23:
The jointMotorTorque only reports the applied torque in POSITION or VELOCITY control modes, since it is implemented by reporting the torque applied by the constraint.

If you use TORQUE control, you provide the actual torque that is applied yourself, so you have the values that is applied already.

avikde, 2017-12-23:
Makes sense, thanks!

closed 2017-12-23 by avikde

erwincoumans, 2017-12-23:
I clarified it in the getJointState documentation of the [PyBullet quickstart guide](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.p3s2oveabizm).

erwincoumans, 2017-12-23:
By the way, we'll implement a PD controller soon, that is automatically applied right before each stepSimulation call. Currently we implemented the PD controller together with motor model for Minitaur in Python here:

https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_envs/bullet/motor.py