bulletphysics/bullet3 #1488: Getting contact forces/wrenches between two bodies

avikde, 2017-12-29:
Is it possible to get the interaction forces between two bodies that are making contact? (e.g. a force if there is a sliding or pinned contact, full wrench if the angle is constrained as well)

I looked at:
* https://github.com/bulletphysics/bullet3/issues/406 but it's not totally clear to me if one of the referenced examples is what I should look at
* In the pybullet quickstart guide, I saw `enableJointForceTorqueSensor` and (I assume) its equivalent in the C API, `b3CreateSensorEnable6DofJointForceTorqueSensor` -- this works, but of course can only be applied to a joint in the urdf (as far as I can tell), and not to a new contact point

erwincoumans, 2017-12-30:
You can use pybullet.getContactPoints to access the contact points computed/used during the last call to stepSimulation. Each contact has an normalForce field that you can multiply with the contactNormal to get the normal interaction force. The friction forces are not exposed yet at the moment, but we can add them if needed.

avikde, 2017-12-30:
Thanks, that seems to work, but could you confirm that I am using it correctly (pull request #1490)? For your convenience I added a few functions to the RobotSimulator example, but I have three questions:
* Logically, it seems incorrect to be providing a joint ID (vs a link in the URDF) to this function... but in the pybullet guide it seems to suggest using a joint ID so that's what I used
* Often the number of contacts even when filtering by the link index is higher than expected and these multiple contacts have similar normal forces... do you have any tips how to disambiguate (or understand what the source is)?
* In the RobotSimulator example, I am printing the contact forces, and the net z force seems quite far from the weight of the body, and especially if I pick the body and move it with my mouse, the result changes quite a bit. Also, the distribution between the 4 legs seems more varied than I'd expect. I thought @erwincoumans you had an example of this already -- hopefully you can point out my mistake.

Thanks!

erwincoumans, 2018-01-01:
I'll recreate a pybullet test first and look at it in a few days. We get very reliable contact forces that sum up well for our Minitaur experiments. Happy 2018!

erwincoumans, 2018-01-10:
(copy from comment in PR)
Here is a better implementation. I also [updated the MinitaurSetup](https://github.com/bulletphysics/bullet3/pull/1503) to allow switching to more up-to-date Minitaur simulation models. I consistently get 73.2 Newton for newer minitaur and 60 Newton for older model (which relates to approx 7.3kg and 6kg)

~~~~~~
 int b3RobotSimulatorClientAPI::getContactNormalForce(int bodyA, int linkA, int bodyB, int linkB, b3Vector3* normalForceOut)
 {
 	b3SharedMemoryCommandHandle commandHandle = b3InitRequestContactPointInformation(m_data->m_physicsClientHandle);
 	b3SetContactFilterBodyA(commandHandle, bodyA);
 	b3SetContactFilterLinkA(commandHandle, linkA);
 	b3SetContactFilterBodyB(commandHandle, bodyB);
 	b3SetContactFilterLinkB(commandHandle, linkB);
 
 	b3SharedMemoryStatusHandle statusHandle = b3SubmitClientCommandAndWaitStatus(m_data->m_physicsClientHandle, commandHandle);
 	int statusType = b3GetStatusType(statusHandle);
 	if (statusType == CMD_CONTACT_POINT_INFORMATION_COMPLETED)
 	{
 		b3ContactInformation contactInfo;
 		b3GetContactPointInformation(m_data->m_physicsClientHandle, &contactInfo);
 		// printf("nc: %d\n", contactInfo.m_numContactPoints);
 		
		normalForceOut->x = 0;
 		normalForceOut->y = 0;
 		normalForceOut->z = 0;
 		for (int i=0;i<contactInfo.m_numContactPoints;i++)
 		{
 			b3ContactPointData* c = &contactInfo.m_contactPointData[i];
 			normalForceOut->x += c->m_contactNormalOnBInWS[0] * c->m_normalForce;
 			normalForceOut->y += c->m_contactNormalOnBInWS[1] * c->m_normalForce;
 			normalForceOut->z += c->m_contactNormalOnBInWS[2] * c->m_normalForce;
 		}
		return 0;
 	}
 	return -1;
 }
  ~~~~~~

erwincoumans, 2018-01-11:
[toeforce.zip](https://github.com/bulletphysics/bullet3/files/1624429/toeforce.zip)
Attached is a diff to enable toe forces.
If the Minitaur lands, there is a symmetry, as we expect. If you drag the mouse and let loose, the leg friction with ground will cause assymetry, also as expected.

carismoses, 2019-04-04:
Any progress on exposing the frictional interaction forces? I am also interested in getting the full reaction force applied to a link (that may not necessarily be attached to any joints), so I cannot use the enableJointForceTorqueSensor() option.

erwincoumans, 2019-04-04:
PyBullet.getContactPoints already returns the two lateral friction directions and forces (in addition to the normal direction/force).

closed 2019-04-04 by erwincoumans

carismoses, 2019-04-04:
Okay, thanks! I was wondering what all of the extra output from p.getContactPoints() was. I am using pyBullet. Could you update the documentation so that I know what indices access the lateral information?

erwincoumans, 2019-04-04:
The documentation was already up-to-date long ago, did you check getClosestPoints?
https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.cb0co8y2vuvc

carismoses, 2019-04-04:
Yes, I deleted my comment because I realized that I copied the documentation to my Drive a long time ago and was not looking at the latest version. Sorry!

richardrl, 2021-05-22:
Is the friction force computation exposed in pybullet? How can I extract it?

iLUVE69, 2023-06-28:
what are two lateral friction direction can anyone explain?