bulletphysics/bullet3 #1497: Collision filtering for pybullet?

benelot, 2018-01-07:
For some other experiments I am doing with pybullet I wanted to ask if there is some access to collision filtering I am not aware of. So If I load two robots into one world, but do not make them collide with each other, I am under the impression that I can do faster simulation than if I load them in one after the other (maybe I should verify that at some time). Anyhow, I wanted to do the same in pybullet, but did not see any method to do this, correct? Or is there a way to use multiple physics servers at the same time in pybullet and quickly switch my ExampleBrowser view from one to the other?

erwincoumans, 2018-01-10:
You can create multiple physics servers (DIRECT) but not easily visualize (only one GUI at a time). One way is to extract all data and then synchronize the state using the saveBullet/loadState combination (if the worlds have the identical setup). I'll work on a way to easily visualize a 'DIRECT' simulation.

No access to collision filtering in PyBullet at the moment.

benelot, 2018-01-10:
Ok thanks! I can work with that.