bulletphysics/bullet3 #1504: Unable to load color for MJCF files

ethanluoyc, 2018-01-10:
Hi,

It appears that when using the `pybullet` API, the colour for the models in MJCF files are not loaded properly. Right now, the colors are somehow picked randomly. I am not sure how this can be fixed.

To reproduce, simply run the examples. If I change the model files in mjcf folder, there will not be any effect if I change the color.

Thanks!

benelot, 2018-01-11:
It is not extracted for now. And they are indeed picked randomly. That is
all I know.

ethanluoyc, 2018-01-11:
@benelot That's good to know.

After some digging, it appears that the importing happens here https://github.com/bulletphysics/bullet3/blob/master/examples/Importers/ImportMJCFDemo/BulletMJCFImporter.cpp#L709. It appears that rgba has been parsed. I am not entirely sure why after this the color is not set up. Maybe I am missing something. If someone can point me to the right direction I can probably make a PR and get this fixed.

erwincoumans, 2018-01-11:
At the moment, the MJCF parser doesn't create visual shapes. This means, there is an autogeneration of visual shapes, using the Bullet collision shapes that don't have color.

I have some early patch that creates visual shapes, with color, but don't have texture coordinates, so it doesn't look nice.

Also, I have an early patch that adds planar reflection, but it needs to use the stencil buffer.

Until then, you can use 
~~~~~~~~~~
p.changeVisualShape(sphere,linkIndex,rgbaColor=[red,green,blue,1])
~~~~~~~~~~
for each link, to assign colors.

ethanluoyc, 2018-01-11:
@erwincoumans Thanks! I am working with RL with image observations of the environment. To me, the texture becomes less important but they have to be distinct from the environment. Would you be able to point to me where the Google colours are instantiated in the codebase? I am trying to understand the how pybullet works right now. I will like to be able to extend it in the long term :D

erwincoumans, 2018-01-12:
See this pull request, once merged, use the option 
~~~~~~~~~
 p.loadMJCF(filename, flags=p.URDF_MJCF_COLORS_FROM_FILE
~~~~~~~~~
When using the option, it uses a highly tesselated spheres/capsules without instancing. We can optimize that later.

ethanluoyc, 2018-01-12:
Thanks! @erwincoumans

closed 2018-01-24 by erwincoumans