bulletphysics/bullet3 #1505: Load urdf returns an id but nothing has been rendered in the simulation

k-maheshkumar, 2018-01-11:
I have following lines in a class,
```
 location_prefix = '/home/bullet3/data/'
urdf_file = location_prefix + "kuka_iiwa/model.urdf"
  self.table_id = self.place_items_from_urdf(urdf_file=location_prefix + "table/table.urdf",
                                                   position=[0, 0, 0.0])
 pland_id = self.place_items_from_urdf(urdf_file=location_prefix + "plane.urdf",
                                              position=[0, 0, 0.0])
self.robot_id = self.place_items_from_urdf(urdf_file, position=[0, 0.25, 0.6])

 def place_items_from_urdf(self, urdf_file, position, orientation=None, use_fixed_base=True):

        if orientation is None:
            urdf_id = sim.loadURDF(urdf_file, basePosition=position, useFixedBase=use_fixed_base)
        else:
            urdf_id = sim.loadURDF(urdf_file, basePosition=position,
                                   baseOrientation=orientation, useFixedBase=use_fixed_base)
        print urdf_id, urdf_file
        return urdf_id
 def run_simulation(self):
        while 1:
           pass

if __name__ == "__main__":
    sim1 = Simulation()
    sim1.run_simulation()

```
following were printed when I run the code
0 /home/bullet3/data/plane.urdf
1 /home/bullet3/data/table/table.urdf
2 /home/bullet3/data/kuka_iiwa/model.urdf

case1:
When I run the local file, which executes the __main__ function, everything works fine. 

case2:
When I instantiate the class from another class, only plane.urdf got loaded into the simulation and nothing else is loaded.

The above output is same for both the cases. Also, in case 2, queries like sim.getNumJoints(self.robot_id)  works fine. 

Can you please help me addressing this problem.
Thanks in advance.

erwincoumans, 2018-01-11:
I have never seen an issue like this. Please provide a full working script (not just a snippet) that can be executed using plain Bullet. 

Thanks!
Erwin

k-maheshkumar, 2018-01-11:
[bullet_issue.zip](https://github.com/bulletphysics/bullet3/files/1623680/bullet_issue.zip)

I have created an example replicating the issue.

steps to reproduce,
1. Run main.py from scripts directory 
2. click start simulation 

only collision shape will be created and loaded into the simulation, also note that the number of joints printed as 7

But if you run thes cripts/simulation/SimulationWorld.py, everything works as expected.

erwincoumans, 2018-01-11:
I cannot run main.py since I don't have Qt. Try removing Qt and see if the problem is still there. You can try running a separate PyBullet server and connect through SHARED_MEMORY instead (either run the Example Browser in physics server mode, or run the script 
~~~~~~~~
python -m pybullet_envs.examples.runServer
~~~~~~~~

k-maheshkumar, 2018-02-01:
I just found that, everthying works as expected when instantiating the simulationWorld before,
     app = QtGui.QApplication(sys.argv)
but not doesn't work, after the above statement.

erwincoumans, 2018-02-01:
Please use plain pybullet, we don't give Qt support.

closed 2018-02-01 by erwincoumans