bulletphysics/bullet3 #1507: pybullet: unwanted convex hull collision check for obj-file with convex objects inside

airballking, 2018-01-12:
Dear Bullet developers,

I want to have collision checking for a concave mesh (a mug) in pybullet. Following a hint from the documentation, I created a convex decomposition using the tool v-hacd and saved the result as an obj-file. The decomposed mesh looks fine in Blender. However, when I load the mesh through a simple URDF in Bullet, it seems that there still a convex hull is used for the object. 

Mesh in Blender:
[image]

Mesh in Bulllet:
[image]

My questions:
1) Does Pybullet support collision checking for meshes with openings like my convex decomposition of the mug?
2) If yes, could you point me to an example that show-cases this behavior?
3) Also, are there some known pitfalls when using v-hacd to prepare obj-files for Pybullet?

Thank you for your support! Best,
Georg.

P.S.: Here is a small ROS package with the meshes and a dummy application playthings.py that you can use to load my obj-mesh in Bullet: 
[my_pkg.zip](https://github.com/bulletphysics/bullet3/files/1626140/my_pkg.zip)

erwincoumans, 2018-01-12:
Your VHACD obj still contains the original cup, just remove it and it should work fine.
Yes, various VHACD examples are in Bullet/data for example duck_vhacd.obj

[image]

airballking, 2018-01-12:
Thank you very much for the quick and very helpful response. I was not aware that I had to remove the old mesh. I thought VHACD would just replace it.

Also, thanks for adding a cup to your database. That is very nice and helpful.

I consider my question answered. Hence, I close this issue.

closed 2018-01-12 by airballking

a-z-e-r-i-l-a, 2020-01-30:
> Instead of using Blender, you can compile the Bullet SDK using premake and there is an executable that creates VHACD from obj to obj called Bullet/bin/test_vhacd_gmake_x64_release.

@erwincoumans Hi, can you please tell a bit more how we can do this? I'm not sure what I should download or what command to use. Does it also work on stl files or only files with .obj format.

Thanks

erwincoumans, 2020-03-16:
@a-z-e-r-i-l-a PyBullet 2.6.7 onwards comes with vhacd included

~~~~~~~~~~~~~~~
pip3 install pybullet --upgrade --user

#now in a script you can use

import pybullet as p
import pybullet_data as pd
import os

p.connect(p.DIRECT)
name_in = os.path.join(pd.getDataPath(), "duck.obj")
name_out = "duck_vhacd2.obj"
name_log = "log.txt"
p.vhacd(name_in, name_out, name_log, alpha=0.04,resolution=50000 )
~~~~~~~~~~~~~~~

francogassibe, 2023-07-24:
Hi there, i use the v-hacd tool and in general i have no problems, but recently i had problems with some meshes. 
Do someone know what requirements does my mesh needs to have? 
Im thinking about things like:
- if its a closed mesh
- the size of the mesh (if its too heavy)
- connectivity

The error i get is just "b3Printf: Request createCollisionShape failed"

My .stl files come from a Solid Works assembled project so i think that there might be some screws or  litle parts inside my main part that may cause problems. 

Here is one of my STLs [my_piece.zip](https://github.com/bulletphysics/bullet3/files/12145466/my_piece.zip)
Thank you in advance