bulletphysics/bullet3 #1510: Build errors in the python module

yurivict, 2018-01-15:
Build of the py27-bullet3-2.87 FroeBSD port fails with clang:
```
cc -DNDEBUG -O2 -pipe -fstack-protector -isystem /usr/local/include -fno-strict-aliasing -isystem /usr/local/include -fPIC -Isrc -Iexamples/ThirdPartyLibs -Iexamples/ThirdPartyLibs/Glew -Iexamples/ThirdPartyLibs/enet/include -Iexamples/ThirdPartyLibs/clsocket/src -I/usr/local/include/python2.7 -c examples/SharedMemory/PhysicsServerExample.cpp -o build/temp.freebsd-12.0-CURRENT-amd64-2.7/examples/SharedMemory/PhysicsServerExample.o -DGWEN_COMPILE_STATIC -DBT_USE_DOUBLE_PRECISION -DBT_ENABLE_ENET -DBT_ENABLE_CLSOCKET -DB3_DUMP_PYTHON_VERSION -D_BSD -I/usr/X11R6/include -DHAS_SOCKLEN_T
examples/SharedMemory/PhysicsServerExample.cpp:2419:5: error: non-constant-expression cannot be narrowed from type 'double' to 'float' in initializer list [-Wc++11-narrowing]
                                m_multiThreadedHelper->m_userDebugText[i].m_textColorRGB[0],
                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
examples/SharedMemory/PhysicsServerExample.cpp:2419:5: note: insert an explicit cast to silence this issue
                                m_multiThreadedHelper->m_userDebugText[i].m_textColorRGB[0],
                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                static_cast<float>(                                        )
examples/SharedMemory/PhysicsServerExample.cpp:2420:5: error: non-constant-expression cannot be narrowed from type 'double' to 'float' in initializer list [-Wc++11-narrowing]
                                m_multiThreadedHelper->m_userDebugText[i].m_textColorRGB[1],
                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
examples/SharedMemory/PhysicsServerExample.cpp:2420:5: note: insert an explicit cast to silence this issue
                                m_multiThreadedHelper->m_userDebugText[i].m_textColorRGB[1],
                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                static_cast<float>(                                        )
examples/SharedMemory/PhysicsServerExample.cpp:2421:5: error: non-constant-expression cannot be narrowed from type 'double' to 'float' in initializer list [-Wc++11-narrowing]
                                m_multiThreadedHelper->m_userDebugText[i].m_textColorRGB[2],
                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
examples/SharedMemory/PhysicsServerExample.cpp:2421:5: note: insert an explicit cast to silence this issue
                                m_multiThreadedHelper->m_userDebugText[i].m_textColorRGB[2],
                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                static_cast<float>(                                        )
examples/SharedMemory/PhysicsServerExample.cpp:2424:20: error: non-constant-expression cannot be narrowed from type 'double' to 'float' in initializer list [-Wc++11-narrowing]
                        float pos[3] = {m_multiThreadedHelper->m_userDebugText[i].m_textPositionXYZ1[0],
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
examples/SharedMemory/PhysicsServerExample.cpp:2424:20: note: insert an explicit cast to silence this issue
                        float pos[3] = {m_multiThreadedHelper->m_userDebugText[i].m_textPositionXYZ1[0],
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                        static_cast<float>(                                            )
examples/SharedMemory/PhysicsServerExample.cpp:2425:4: error: non-constant-expression cannot be narrowed from type 'double' to 'float' in initializer list [-Wc++11-narrowing]
                        m_multiThreadedHelper->m_userDebugText[i].m_textPositionXYZ1[1],
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
examples/SharedMemory/PhysicsServerExample.cpp:2425:4: note: insert an explicit cast to silence this issue
                        m_multiThreadedHelper->m_userDebugText[i].m_textPositionXYZ1[1],
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                        static_cast<float>(                                            )
examples/SharedMemory/PhysicsServerExample.cpp:2426:4: error: non-constant-expression cannot be narrowed from type 'double' to 'float' in initializer list [-Wc++11-narrowing]
                        m_multiThreadedHelper->m_userDebugText[i].m_textPositionXYZ1[2]};
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
examples/SharedMemory/PhysicsServerExample.cpp:2426:4: note: insert an explicit cast to silence this issue
                        m_multiThreadedHelper->m_userDebugText[i].m_textPositionXYZ1[2]};
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                        static_cast<float>(                                            )
6 errors generated.
```

The clang version is likely 5.0.

Also, please remove all ```register``` keywords. They do nothing with modern compiler, only produce warnings:
```
examples/ThirdPartyLibs/BussIK/LinearR3.h:1539:32: warning: 'register' storage class specifier is deprecated and incompatible with C++17 [-Wdeprecated-register]
inline LinearMapR3 operator* ( register double b, const LinearMapR3& A)
                               ^~~~~~~~~
examples/ThirdPartyLibs/BussIK/LinearR3.h:1548:2: warning: 'register' storage class specifier is deprecated and incompatible with C++17 [-Wdeprecated-register]
        register double bInv = 1.0/b;
        ^~~~~~~~~
examples/ThirdPartyLibs/BussIK/LinearR3.h:1554:46: warning: 'register' storage class specifier is deprecated and incompatible with C++17 [-Wdeprecated-register]
inline LinearMapR3& LinearMapR3::operator/= (register double b)
                                             ^~~~~~~~~
```

yurivict, 2018-05-06:
@erwincoumans Any chance to fix this? The build fails with clang-6.

erwincoumans, 2018-05-06:
If you care, fork the repo, fix the error, create a pull request.

yurivict, 2018-05-06:
> If you care, fork the repo, fix the error, create a pull request. Only deal with the error, no other changes in the pull request makes it more like to be accepted.

I am a maintainer of the FreeBSD port. I don't really use it, and can't spend a significant time debugging the third party packages. You as a project owner should be more interested in your project being buildable on various systems with different compilers.

Currently it fails with C++ errors with clang-6.

jslee02, 2018-07-10:
I believe the original issue is fixed by 9ffb05e. But I've encountered similar [build errors but on different code (`InvertedPendulumPDControl.cpp`) on Bionic with clang-6.0](https://travis-ci.org/bulletphysics/bullet3/jobs/402060359#L4432). Let me try to fix it by applying the same approach of 9ffb05e in https://github.com/bulletphysics/bullet3/pull/1788.

commit c254768d (2018-07-19): Merge pull request #1793 from jslee02/fix_build_clang6

Fix build errors with Clang 6 (also resolve #1510)
closed 2018-07-19 by erwincoumans