bulletphysics/bullet3 #1514: Falling cube unexpectedly spins on colliding with plane shape

groud, 2018-01-16:
Hi! 

I am from the Godot community, and we faced a strange behaviour with a simple cube and a plane :
https://github.com/godotengine/godot/issues/15770

When it falls on the ground (an infinite plane collider), the cube goes a little bit through the ground and gets out with significant spin. We wonder if this is an expected behaviour. The problem does not happen when a box is used for the ground shape.

A video of the problem is uploaded [here](https://github.com/godotengine/godot/files/1633889/Bullet.vs.Godot.physics.flv.zip).

Thanks !

erwincoumans, 2018-01-16:
By default, Bullet only computes 1 point at a time for convex versus btPlaneShape, and it adds more points to the cache over multiple simulation steps. So it is expected behaviour. It is easy to improve, if it is very important. How does the comparison look like when using a single triangle as ground?

groud, 2018-01-16:
It is not that important, as it works fine with a single triangle. :)
I am not sure it is worth improving it. If it's done, it obviously would be a good news, but to be honest the only good reason for it on our side would be to avoid users opening issues about it (as the behaviour looks a little bit unintuitive). But it should be fine if we document this clearly.

Thank you for you time.

closed 2021-04-26 by erwincoumans