bulletphysics/bullet3 #1517: Unknown argument for "changeVisualShape" for the pybullet API

ethanluoyc, 2018-01-16:
In the pybullet documentation for the function `changeVisualShape` some mysterious argument named `shapeIndex` is mentioned. What is that argument for? The doc mentioned referring to `getVisualShapeData`. However, I couldn't find returned output with the same name from the doc.

Also is it possible to change the shape of the object on the fly? i.e. say I would like to suddently
scale a sphere in the rendering. `changeVisualShape` does not seem to have arguments for this.

avikde, 2018-01-21:
I recently had the same problem, and (albeit in the C API), did something like this to find the shape ID

```cpp
	// Loop through all joints
	for (int i=0; i<numJoints; ++i) {
		b3GetJointInfo(kPhysClient, b3id, i, &jointInfo);
		if (jointInfo.m_jointName[0]) {
			jointNameToId[std::string(jointInfo.m_jointName)] = i;
			printf("jointId=%d -> %s\n", jointInfo.m_jointIndex, jointInfo.m_jointName);
		} else {
			continue;
		}
	}

// now loop through all the shapes
	b3VisualShapeInformation vsi;
	b3VisualShapeData *vsd;
	command = b3InitRequestVisualShapeInformation(kPhysClient, BODY_ID);
	statusHandle = b3SubmitClientCommandAndWaitStatus(kPhysClient, command);
	b3GetVisualShapeInformation(kPhysClient, &vsi);
	printf("numshapes = %d\n", vsi.m_numVisualShapes);
	for (int i=0; i<vsi.m_numVisualShapes; ++i)
	{
		vsd = &vsi.m_visualShapeData[i];
		// m_linkIndex in this corresponds to the jointIndex above. 
		printf("shapeId=%d -> jointId=%d\n", i, vsd->m_linkIndex);
	}
```

By cross-referencing those, hopefully you have the right shape ID. I'm not confident this is right (especially since it's a little confusing to conflate linkIndex and jointIndex -- which the pybullet guide says to do on occasion).

erwincoumans, 2018-01-21:
>> changeVisualShape some mysterious argument named shapeIndex is mentioned.

It is best to ignore this optional argument, since it is only experimental for some internal experiments in some of our project and not fully implemented (leave shapeIndex out or use -1). I updated the quickstart guide to explain this. This 'shapeIndex' is a way to identify which of the visual shapes you want to modify. Since URDF can have more than one (or zero) visual shapes per link, just an object unique id and link (=joint) index is not sufficient.

@avikde: linkIndex == jointIndex, since each link has exactly one joint connecting it to the parent. We used linkIndex and jointIndex exchangeably since they are the same.

closed 2018-01-21 by erwincoumans